realsenseai/realsense-ros #832: Error compiling on Intel aero after update

fervanegas, 2019-07-01:
Hi,
After updating the library, I am having this issue when I try to compile. It has been working fine but after upgrading (sudo apt-get update and then sudo apt-get upgrade) I got errors and tried to recompile but I am getting this error now:
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - realsense2_description
-- ~~  - realsense2_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense2_description'
-- ==> add_subdirectory(realsense-ros/realsense2_description)
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/realsense2/realsense2Targets.cmake:107 (message):
  The imported target "realsense2::uvc_fw" references the file

     "/usr/lib/x86_64-linux-gnu/libuvc_fw.a"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib/x86_64-linux-gnu/cmake/realsense2/realsense2Targets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/realsense2/realsense2Config.cmake:35 (include)
  realsense-ros/realsense2_camera/CMakeLists.txt:37 (find_package)

rgreid, 2019-07-01:
We have the same issue on a couple of computers here. Quick fix is to comment out references to `libuvc_fw` in these CMake scripts:  
* `/usr/lib/x86_64-linux-gnu/cmake/realsense2/realsense2Targets.cmake`
* `/usr/lib/x86_64-linux-gnu/cmake/realsense2/realsense2Targets-release.cmake`

tykurtz, 2019-07-01:
Ran into this as well. Pinning an older release worked for me.

> Modified from https://github.com/IntelRealSense/realsense-ros/issues/502#issuecomment-433464096
```sh
version="2.23.0-0~realsense0.1166"
sudo apt-get install librealsense2-dkms
sudo apt-get install librealsense2=${version}
sudo apt-get install librealsense2-gl=${version} 
sudo apt-get install librealsense2-utils=${version}  
sudo apt-get install librealsense2-dev=${version}
sudo apt-get install librealsense2-dbg=${version}
```

fervanegas, 2019-07-02:
Thanks a lot @rgreid that solved the issue.

rgreid, 2019-07-02:
Can I suggest you re-open this? 

Package pinning and editing CMake scripts aren't good solutions. 

We're seeing this on Ubuntu 16.04 and 18.04 after upgrading `librealsense2` to version `2.24.0-0~realsense0.1239`

SteveGrehl, 2019-07-02:
+1 for opening the issue again

I run into the same issue, with a different machine tho, but still. I Think this is relevant to the developers and the community. Since it is bad practice to open new issues with the same topic, this issue should stay open until its properly solved.

rgreid, 2019-07-02:
We've had this issue on a laptop and desktop PC also. Perhaps consider changing the issue title to **Error compiling after upgrading librealsense2 to version 2.24.0-0~realsense0.1239**?

doronhi, 2019-07-03:
There was an update in the debian package. 
Try updating librealsense2 (`apt-get upgrade librealsense2-dev`). 
Make sure you get build 0.1243:
```
...
Setting up librealsense2-dev:amd64 (2.24.0-0~realsense0.1243) ...
...
```

SteveGrehl, 2019-07-03:
> There was an update in the debian package.
> Try updating librealsense2 (`sudo apt update && sudo apt upgrade`).
> Make sure you get build 0.1243:
> 
> ```
> ...
> Setting up librealsense2-dev:amd64 (2.24.0-0~realsense0.1243) ...
> ...
> ```
The update fixes this specific issue for me. I got a compile error tho, I will look into it first and open another issue if relevant.

Have you been able to build the `realsense-ros` package?

rgreid, 2019-07-04:
Thanks! With `2.24.0-0~realsense0.1243` installed and the cmake scripts not edited ([above](https://github.com/IntelRealSense/realsense-ros/issues/832#issuecomment-507163200)), the `realsense2_camera` ROS package builds and runs fine. Tested with T265 and D435 cameras.

doronhi, 2019-07-07:
Great to hear! Can we close the issue then? @fervanegas ?

fervanegas, 2019-07-07:
Yes, All good I tested as well.

closed 2019-07-07 by fervanegas

cjue, 2019-07-31:
This issue still exists in the current aarch64 binary package, version `2.24.0-5ubuntu5~1.gbpf112ed`.

Commenting out the reference to libuvc_fw.a as mentioned by @rgreid fixes the CMake error: https://github.com/IntelRealSense/realsense-ros/issues/832#issuecomment-507163200

doronhi, 2019-08-06:
The bug was fixed before for the X86. I passed it over to librealsense's team to check the build for arm.

ev-mp, 2019-08-07:
@cjue , thanks for pointing this out.
The inconsistency in the `librealsense2-dev` package for arm64 was fixed in the latest release.
Note that this requires you to update Librealsense to v2.25 and then rebuild against this version.

cjue, 2019-08-07:
Thanks for the update!

jajosheni, 2019-12-20:
I want to open this issue again. I think there is the same bug with the new update.

doronhi, 2019-12-30:
> I want to open this issue again. I think there is the same bug with the new update.

Please provide the versions you are trying to build, the machine, kernel, log files and everything else that seems relevant. Consider opening a new issue as with different versions than specified here it is another issue, I think.