realsenseai/realsense-ros #835: Realsense D435 and D435i detected with lower firmware but not working

AnthonyKyung11, 2019-07-02:
Hi, I didn't see any other issue with the same problem I'm having so I'm making one.
I'm using librealsense2 , build 2.22.0  in ubuntu 16.04 in the Intel Aero, with realsense-ros  2.2.6.
The problem I have is with the realsense cameras D435 and D435i whit firmware 5.10.3 to 5.11.6.250.
If I connect the cameras (individually) with the Firmware 5.11.6.250 the D435i does not gets recognized with rs-enumerate-devices and the D435 gets recognized but it gives me the following error when running roslaunch realsense2_camera rs_aligned_depth.launch:

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: /home/aero/sandbo...
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
[... 216 lines cut ...]
 02/07 18:21:50,165 ERROR [140004231726848] (types.h:243) Cannot open '/dev/video0 Last Error: No such file or directory
 02/07 18:21:50,166 ERROR [140004231726848] (global_timestamp_reader.cpp:152) Error during time_diff_keeper polling: Cannot open '/dev/video0 Last Error: No such file or directory
 02/07 18:21:51,036 ERROR [140004423952128] (sensor.cpp:260) An error has occurred while stop_streaming()!
 02/07 18:21:51,037 ERROR [140004423952128] (types.h:243) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 02/07 18:21:51,037 ERROR [140004423952128] (sensor.cpp:260) An error has occurred while stop_streaming()!
 02/07 18:21:51,222 WARNING [140005024999296] (types.cpp:48) stop_streaming() failed. UVC device is not streaming!

[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
[1mdone[0m

---------------------------------------------------------------------------------------------------------------------

I tried the following firmwares for both cameras, but the only one that detects the cameras is the 5.10.3:
-5.10.6
-5.10.13
-5.11.1
-5.11.4
-5.11.6.250

Do you have any idea what could be causing this problem for both cameras?

forwardkth, 2019-08-24:
I have the same issue. The only working firmware with Jetson TX2 is 5.10.3. The D435I can not be detected if using other firmwares.

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Is this issue still happening? I can confirm firmware latest 5.11.15 with RealSense SDK build 2.29 work on D435/D435i.
Please let us know if you have further question question on this topic. Thank you.

closed 2019-11-15 by RealSenseCustomerSupport