realsenseai/realsense-ros #836: Option to record in Realsense viewer bag file format

anug7, 2019-07-03:
Currently, the rosbag file stored by realsense viewer has enough meta information to do post processing like alignment. _Is there a similar option to store bag file with realsense-ros wrapper so that post processing (like aligning depth frame with color) can be done?_

If I use option "align_depth" in ros, the frame rate drops considerably. In our case, we need to use ros wrapper to store data.

doronhi, 2019-07-07:
I'm not sure I understand.
What I do understand is the followings:
1. You want to save your data using `rosbag` and not `realsense-viewer` or `rs-record`
2. You will later want to play the bag files using realsense2_camera and do post processing.

If that is the case, I'm afraid it's not straight forward. The rosbag file stored by realsense-viewer is not 100% rosbag file. It is readable by librealsense2 and that's how realsense2_camera uses it. The wrapper - realsense2_camera, cannot play or process rosbag files.

Or, I did I misunderstood the question?

anug7, 2019-07-07:
@doronhi, You understood my question correclty.. If I understand your response correctly, there's no option in ros wrapper to store the data in the format that can be used to do post processing (alignment) with rs-align in librealsense2.
Is that correct? But why there's no option to record the data (like rs-record does) in ros wrapper? this limits the ros wrapper to publish the data & as well as record for any post processing...

doronhi, 2019-07-07:
The idea is that it's product is in ROS topics, and if you want to save those, there are ROS tools for that.
If you want to save the original data and try running realsense2_camera with all sorts of post process configurations, you could save with realsense-viewer or rs-record and play those file with realsense2_camera.
I don't remember the need to save the original data using the wrapper ever came up before.

anug7, 2019-07-07:
We need to have both for our case. Any pointers on where I can start to work on that?

doronhi, 2019-07-08:
I guess you should convert from ROS rosbag file to librealsense2 "rosbag" file. Then you could record with `rosbag record`, convert and play with realsense2_camera.
I believe the 2 formats are not that different but I haven't got around to check it.

anug7, 2019-07-09:
The topics need to record in ROS rosbag format so that librealsense tools can be used for post processing are more (while looking at the **_rs-record_** bag file) and it's not clearly defined..

And the formats are different. There are some additional topics in **_rs-record_** bag file (like file version) which is not published by realsense-ros.

Thank you for your support. I will try to add what **_rs-record_** does in the realsense-ros

anug7, 2019-07-22:
@doronhi , As I mentioned, I created a separate ros package which uses librealsense API to store the bag in rs-record format. It works fine without any issues when I run this node alone. But when I run realsense-ros node for the second Realsense camera (T265), the T265 node doesn't start and struct at LIBUSB error. Please find attached messages.

`
19:11:25.712 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.722 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.732 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.743 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.753 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.763 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.773 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.783 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.793 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.804 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.814 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.824 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.834 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.844 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.854 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.864 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.875 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.885 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.895 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.905 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.915 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.926 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.936 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.946 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.956 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.966 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.977 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.987 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:25.997 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:26.007 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
19:11:26.017 [2570] [E] Device-3310: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
`

Setup Details:
1. Camera1:  D435 camera - Custom node to store data in rs-record format. 
2. Camera2: T265 camera - Intel Realsense-ros node

anug7, 2019-07-30:
@doronhi , Any update on this?

doronhi, 2019-07-30:
Currently, there is a problem running T265 on 2 separate nodes.
I expect the problem is that your first node takes hold of the T265 in it's context, thus prevent the other node from using it.
Until this is solved, realsense2_camera have a logic for releasing the T265 from the context if it is not using it, so that another node will be able to claim it.
Check the usage of the function unload_tracking_module() referenced [here](https://github.com/IntelRealSense/realsense-ros/blob/5ef502ebcee5508adc073d228b4ed52d07acd5f7/realsense2_camera/src/realsense_node_factory.cpp#L80)
You may need a similar logic in your code.

kroq-gar78, 2019-09-03:
@anug7 Do you have the code for your module anywhere? I am only using D400 cameras, so that bug shouldn't affect me.

anug7, 2019-09-03:
@kroq-gar78 , the code is very much tied to D435 camera. I am not sure whether that can be useful to you. 

Anyways, I am attaching code [here](https://gist.github.com/anug7/b4ec25632cd2859109a306a71fa98f13)

closed 2024-03-29 by MartyG-RealSense