realsenseai/realsense-ros #837: how to get infra1 and infra2 image using cv_bridge in ROS?

npd26, 2019-07-04:


doronhi, 2019-07-07:
In python it will look something like that:
```
    sub = rospy.Subscriber('/camera/infra1/image_rect_raw', msg_Image, image_callback, queue_size=1)

    def image_callback(self, image_msg):
        system_time = time.time()
        try:
            image = self.bridge.imgmsg_to_cv2(image_msg, desired_encoding="passthrough")
        except CvBridgeError as e:
            print(e)
        print image.shape
```

RealSenseCustomerSupport, 2019-11-04:
----------------------------------------------

@npd26 With the reply from doronhi, did you get it through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@npd26 Sorry that we didn't get response from you for weeks. Will close it at this point. If you still have questions/issues, please feel free to create another one. Thanks!

closed 2019-12-02 by RealSenseCustomerSupport