realsenseai/realsense-ros #838: undefined symbol

tonnesfn, 2019-07-05:
Operating System | Ubuntu 18.04 LTS
Kernel | 4.15.0-54-generic
ROS | melodic
ROS RealSense | 2.2.6
librealsense | realsense-1.3.6
D435 Firmware | 05.11.06.250

I have installed the librealsense2 packages, along with realsense-ros and ros-kinetic-ddynamic-reconfigure. Everything works correctly in realsense-viewer, but when I try to run 'roslaunch realsense2_camera rs_camera.launch', I get a symbol lookup error. I assume I have installed an incorrect version of something alone the way. Any suggestions?

 ```
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
* /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [11631]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fb73ea02-9ebf-11e9-b512-d89ef305685d
process[rosout-1]: started with pid [11642]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [11649]
process[camera/realsense2_camera-3]: started with pid [11650]
[ INFO] [1562288301.357499722]: Initializing nodelet with 4 worker threads.
[ INFO] [1562288301.432916106]: RealSense ROS v2.2.6
[ INFO] [1562288301.432964374]: Running with LibRealSense v2.24.0
[ INFO] [1562288301.899968870]: getParameters...
[ INFO] [1562288302.018406344]: setupDevice...
[ INFO] [1562288302.018440396]: JSON file is not provided
[ INFO] [1562288302.018461065]: ROS Node Namespace: camera
[ INFO] [1562288302.018484658]: Device Name: Intel RealSense D435
[ INFO] [1562288302.018503455]: Device Serial No: 828112073073
[ INFO] [1562288302.018519961]: Device FW version: 05.11.06.250
[ INFO] [1562288302.018533667]: Device Product ID: 0x0B07
[ INFO] [1562288302.018550070]: Enable PointCloud: Off
[ INFO] [1562288302.018567419]: Align Depth: Off
[ INFO] [1562288302.018583404]: Sync Mode: Off
[ INFO] [1562288302.018820253]: Device Sensors: 
[ INFO] [1562288302.019010357]: Stereo Module was found.
[ INFO] [1562288302.019060956]: RGB Camera was found.
[ INFO] [1562288302.019107146]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1562288302.019128502]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1562288302.019142082]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1562288302.019159658]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1562288302.019178706]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1562288302.019193873]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1562288302.019356974]: setupPublishers...
[ INFO] [1562288302.021952481]: Expected frequency for depth = 30.00000
[ INFO] [1562288302.050102365]: Expected frequency for infra1 = 30.00000
[ INFO] [1562288302.069123287]: Expected frequency for infra2 = 30.00000
[ INFO] [1562288302.089850191]: Expected frequency for color = 30.00000
[ INFO] [1562288302.108724303]: setupStreams...
[ INFO] [1562288302.109808732]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1562288302.110108038]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ WARN] [1562288302.110214498]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 640, Height: 480, FPS: 30
[ INFO] [1562288302.111221978]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1562288302.112615257]: insert Depth to Stereo Module
[ INFO] [1562288302.112792725]: insert Color to RGB Camera
[ INFO] [1562288302.112835563]: insert Infrared to Stereo Module
[ INFO] [1562288302.120101618]: num_filters: 0
[ INFO] [1562288302.120172218]: SELECTED BASE:Depth, 0
[ INFO] [1562288302.120788195]: RealSense Node Is Up!
[ INFO] [1562288302.120810709]: Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/USERNAME/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE
[camera/realsense2_camera_manager-2] process has died [pid 11649, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/USERNAME/.ros/log/fb73ea02-9ebf-11e9-b512-d89ef305685d/camera- realsense2_camera_manager-2.log].
log file: /home/USERNAME/.ros/log/fb73ea02-9ebf-11e9-b512-d89ef305685d/camera-realsense2_camera_manager-2*.log
```

doronhi, 2019-07-07:
I couldn't reproduce but it seems like you have more than one installation of ddynamic_reconfigure on your machine. realsense-viewer would not care since it's not using it.

tonnesfn, 2019-07-09:
Could not find more than one install of ddynamic_reconfigure, but a complete wipe and reinstall of the system fixed the issue.

closed 2019-07-09 by tonnesfn

araujokth, 2019-08-19:
I also bumped into this error and the problem was that in the steps when installing the realsense2-camera (following the .travis.yml), when you have to do:
```
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
```
I had skipped to do the last step catkin_make install. Just in case someone else bumps into this

pmirabel, 2020-10-16:
Hello!

I'm facing the same issue here, but a manual `clean` of ros_workspace didn't help.  
@doronhi , could you please tell me a little  more about this ?? : 
> I couldn't reproduce but it seems like you have more than one installation of ddynamic_reconfigure on your machine. realsense-viewer would not care since it's not using it.

Thanks a lot !

pmirabel, 2020-10-16:
> Hello!
> 
> I'm facing similar issue here, but a manual `clean` of ros_workspace didn't help.
> Here is my undefined symbol :
> `librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigure16registerVariableIiEEvRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEET_RKN5boost8functionIFvSA_EEES9_SA_SA_`
> Seems indeed linked to ddynamic reconfigure but ... can't figure out how to deal with that.
> 
> @doronhi , could you please tell me a little more about this ?? :
> 
> > I couldn't reproduce but it seems like you have more than one installation of ddynamic_reconfigure on your machine. realsense-viewer would not care since it's not using it.
> 
> Thanks a lot !

If someone encounter this issue : I was indeed compiling in a "not clean, polluted by old build artifact which was using old librealsense version" build environment. But catkin clean is not sufficient, I had to reinstall everything from scratch, don't know why...

snakehaihai, 2021-04-05:
hey im having similar issue on a intel x64 notebook PC running ubuntu 16.04 
The realsense viewer is running with 42 build and it works well
I downloaded the ROS1 version of the 42 supported package. But end up with this error. 
[image]

I tried to uninstall apt remove ddynamic *  and download the 0.2.0 version from github. But encoutner with build error says the register function not compatiable. I searched in the header file. it was there. 

So i have no idea whats going on.

snakehaihai, 2021-04-05:
Ok my problem solved. 

In my file system

There is 
catkin_ws
driver_ws
deply_ws
test_ws

There was one unknown version of ddynamic in test_ws and it was included automatically in bashrc. remove all possible ddynamic solved it.

arif0342, 2021-04-09:
> Ok my problem solved.
> 
> In my file system
> 
> There is
> catkin_ws
> driver_ws
> deply_ws
> test_ws
> 
> There was one unknown version of ddynamic in test_ws and it was included automatically in bashrc. remove all possible ddynamic solved it.

If you don't mind can you please elaborate how to solve this issue? I am facing same problem. Actually I am little bit new to ROS so can you please explain in little detail how to solve this. I am facing this
[image]

snakehaihai, 2021-04-11:
> > Ok my problem solved.
> > In my file system
> > There is
> > catkin_ws
> > driver_ws
> > deply_ws
> > test_ws
> > There was one unknown version of ddynamic in test_ws and it was included automatically in bashrc. remove all possible ddynamic solved it.
> 
> If you don't mind can you please elaborate how to solve this issue? I am facing same problem. Actually I am little bit new to ROS so can you please explain in little detail how to solve this, i will be very much thankful to you. I am facing this
> [image]

i got duplicated build library. Just delete all source in the bashrc and remove all catkin_ws. THen with a clean build and src. then try to build again. This is to make sure that the library is not linked to a wrong version of supporting library

arif0342, 2021-04-15:
> > > Ok my problem solved.
> > > In my file system
> > > There is
> > > catkin_ws
> > > driver_ws
> > > deply_ws
> > > test_ws
> > > There was one unknown version of ddynamic in test_ws and it was included automatically in bashrc. remove all possible ddynamic solved it.
> > 
> > 
> > If you don't mind can you please elaborate how to solve this issue? I am facing same problem. Actually I am little bit new to ROS so can you please explain in little detail how to solve this, i will be very much thankful to you. I am facing this
> > [image]
> 
> i got duplicated build library. Just delete all source in the bashrc and remove all catkin_ws. THen with a clean build and src. then try to build again. This is to make sure that the library is not linked to the wrong version of supporting library ddynamic.
> 
> I tried multiple version to isolate the problem in ddynamic. So for your case, if the solution i said earlier didn`t work out. You can try 43 and 42 and and default ubuntu apt install. see whats the error coming out

Got it, Thank you so much brother

hzlijianjun, 2021-05-19:
I faced similar problem as following:

`/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/jungle/catkin_my_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN14image_geometry18PinholeCameraModelC1Ev`

I solved it by adding the package "image_geometry" to CMakeLists.txt:
[image]

I hope you drawing inspiration from my method!

arif0342, 2021-05-21:
Thanks Got it.

danialdunson, 2022-06-04:
thank you, [snakehaihai](https://github.com/snakehaihai)! 

my ~/.bashrc had second line of `source /opt/ros/melodic/setup.bash` and extraneous lines of `PATH=/home/***/bin:$PATH` that were causing issues.

Akumar201, 2022-10-03:
I followed the above steps , then also getting the same error here is my .bashrc

# ~/.bashrc: executed by bash(1) for non-login shells.
# see /usr/share/doc/bash/examples/startup-files (in the package bash-doc)
# for examples

# If not running interactively, don't do anything
case $- in
    *i*) ;;
      *) return;;
esac

# don't put duplicate lines or lines starting with space in the history.
# See bash(1) for more options
HISTCONTROL=ignoreboth

# append to the history file, don't overwrite it
shopt -s histappend

# for setting history length see HISTSIZE and HISTFILESIZE in bash(1)
HISTSIZE=1000
HISTFILESIZE=2000

# check the window size after each command and, if necessary,
# update the values of LINES and COLUMNS.
shopt -s checkwinsize

# If set, the pattern "**" used in a pathname expansion context will
# match all files and zero or more directories and subdirectories.
#shopt -s globstar

# make less more friendly for non-text input files, see lesspipe(1)
[ -x /usr/bin/lesspipe ] && eval "$(SHELL=/bin/sh lesspipe)"

# set variable identifying the chroot you work in (used in the prompt below)
if [ -z "${debian_chroot:-}" ] && [ -r /etc/debian_chroot ]; then
    debian_chroot=$(cat /etc/debian_chroot)
fi

# set a fancy prompt (non-color, unless we know we "want" color)
case "$TERM" in
    xterm-color|*-256color) color_prompt=yes;;
esac

# uncomment for a colored prompt, if the terminal has the capability; turned
# off by default to not distract the user: the focus in a terminal window
# should be on the output of commands, not on the prompt
#force_color_prompt=yes

if [ -n "$force_color_prompt" ]; then
    if [ -x /usr/bin/tput ] && tput setaf 1 >&/dev/null; then
	# We have color support; assume it's compliant with Ecma-48
	# (ISO/IEC-6429). (Lack of such support is extremely rare, and such
	# a case would tend to support setf rather than setaf.)
	color_prompt=yes
    else
	color_prompt=
    fi
fi

if [ "$color_prompt" = yes ]; then
    PS1='${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
else
    PS1='${debian_chroot:+($debian_chroot)}\u@\h:\w\$ '
fi
unset color_prompt force_color_prompt

# If this is an xterm set the title to user@host:dir
case "$TERM" in
xterm*|rxvt*)
    PS1="\[\e]0;${debian_chroot:+($debian_chroot)}\u@\h: \w\a\]$PS1"
    ;;
*)
    ;;
esac

# enable color support of ls and also add handy aliases
if [ -x /usr/bin/dircolors ]; then
    test -r ~/.dircolors && eval "$(dircolors -b ~/.dircolors)" || eval "$(dircolors -b)"
    alias ls='ls --color=auto'
    #alias dir='dir --color=auto'
    #alias vdir='vdir --color=auto'

    alias grep='grep --color=auto'
    alias fgrep='fgrep --color=auto'
    alias egrep='egrep --color=auto'
fi

# colored GCC warnings and errors
#export GCC_COLORS='error=01;31:warning=01;35:note=01;36:caret=01;32:locus=01:quote=01'

# some more ls aliases
alias ll='ls -alF'
alias la='ls -A'
alias l='ls -CF'

# Add an "alert" alias for long running commands.  Use like so:
#   sleep 10; alert
alias alert='notify-send --urgency=low -i "$([ $? = 0 ] && echo terminal || echo error)" "$(history|tail -n1|sed -e '\''s/^\s*[0-9]\+\s*//;s/[;&|]\s*alert$//'\'')"'

# Alias definitions.
# You may want to put all your additions into a separate file like
# ~/.bash_aliases, instead of adding them here directly.
# See /usr/share/doc/bash-doc/examples in the bash-doc package.

if [ -f ~/.bash_aliases ]; then
    . ~/.bash_aliases
fi

# enable programmable completion features (you don't need to enable
# this, if it's already enabled in /etc/bash.bashrc and /etc/profile
# sources /etc/bash.bashrc).
if ! shopt -oq posix; then
  if [ -f /usr/share/bash-completion/bash_completion ]; then
    . /usr/share/bash-completion/bash_completion
  elif [ -f /etc/bash_completion ]; then
    . /etc/bash_completion
  fi
fi

snakehaihai, 2022-10-05:
> I followed the above steps , then also getting the same error here is my .bashrc
>
> ```
> # ~/.bashrc: executed by bash(1) for non-login shells.
> # see /usr/share/doc/bash/examples/startup-files (in the [...]

your bashrc has no source. you didnt source the ros in /opt and didnt source the ros in the ws. follow ros install and get the following two lines inside
```

echo "source /opt/ros/melodic/setup.bash" >> ~/.bashrc
source ~/.bashrc

```

also if you have workspace do the same