realsenseai/realsense-ros #840: warning on starting T265

gbr1, 2019-07-08:
Hi!
Testing T265 launching:
`roslaunch realsense2_camera rs_t265.launch`
 I notice this warning message:
`[ WARN] [1562626265.620639228]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)`
I don't know if it is an important warning.
I also tested using `enable_sync:=true`.  demo_t265.launch gives me same warning.
I'm running on kinetic, ubuntu 16 and latest release of librealsense (2.24).
Thank you in advance!

doronhi, 2019-07-10:
This warning is meaningless see #803 
As it is annoying I'll try to suppress is.
Thanks @gbr1 :)

doronhi, 2019-07-11:
It is a problem with the way T265 reports it's time domain.
Fixed in [librealsense PR #4403](https://github.com/IntelRealSense/librealsense/pull/4403)

gbr1, 2019-07-11:
@doronhi thank you!

closed 2019-07-11 by doronhi