realsenseai/realsense-ros #842: LIBUSB_ERROR_IO in Raspberry Pi

echoGee, 2019-07-09:
Launching `roslaunch realsense2_camera rs_camera.launch` on a Raspberry pi 3 on code compiled using the master branch causes the follow error after a few minutes. 
The PI is powered using a 15watt power supply to avoid any power related issues 

>  NODES
>   /camera/
>     realsense2_camera (nodelet/nodelet)
>     realsense2_camera_manager (nodelet/nodelet)
> 
> auto-starting new master
> process[master]: started with pid [10141]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 6e477bce-a29a-11e9-a3ef-b827ebb74f65
> process[rosout-1]: started with pid [10154]
> started core service [/rosout]
> process[camera/realsense2_camera_manager-2]: started with pid [10171]
> process[camera/realsense2_camera-3]: started with pid [10172]
> [ INFO] [1562711981.519156607]: Initializing nodelet with 4 worker threads.
> [ INFO] [1562711981.678863376]: RealSense ROS v2.2.6
> [ INFO] [1562711981.679028687]: Running with LibRealSense v2.24.0
> [ INFO] [1562711982.098220428]: No calib_odom_file. No input odometry accepted.
> [ INFO] [1562711982.098691727]: getParameters...
> [ INFO] [1562711982.630289245]: setupDevice...
> [ INFO] [1562711982.630488462]: JSON file is not provided
> [ INFO] [1562711982.630611899]: ROS Node Namespace: camera
> [ INFO] [1562711982.630721169]: Device Name: Intel RealSense T265
> [ INFO] [1562711982.630802106]: Device Serial No: 905312111095
> [ INFO] [1562711982.630886636]: Device FW version: 0.0.18.5715
> [ INFO] [1562711982.630966532]: Device Product ID: 0x0B37
> [ INFO] [1562711982.631043719]: Enable PointCloud: Off
> [ INFO] [1562711982.631117364]: Align Depth: Off
> [ INFO] [1562711982.631192780]: Sync Mode: Off
> [ INFO] [1562711982.631344654]: Device Sensors: 
> [ INFO] [1562711982.631621632]: Tracking Module was found.
> [ INFO] [1562711982.631791735]: (Depth, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1562711982.631880432]: (Color, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1562711982.631962515]: (Infrared, 1) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1562711982.632034858]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1562711982.632105639]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1562711982.632456782]: setupPublishers...
> [ INFO] [1562711982.651546918]: Expected frequency for fisheye1 = 30.00000
> [ INFO] [1562711982.742654240]: Expected frequency for fisheye2 = 30.00000
> [ INFO] [1562711982.845612421]: setupStreams...
> [ INFO] [1562711982.847837563]: fisheye1 stream is enabled - width: 848, height: 800, fps: 30
> [ INFO] [1562711982.849664374]: fisheye2 stream is enabled - width: 848, height: 800, fps: 30
> [ INFO] [1562711982.850174840]: insert Fisheye to Tracking Module
> [ INFO] [1562711982.850457702]: insert Fisheye to Tracking Module
> [ INFO] [1562711982.850579785]: insert Gyro to Tracking Module
> [ INFO] [1562711982.850673222]: insert Accel to Tracking Module
> [ INFO] [1562711982.850757700]: insert Pose to Tracking Module
> [ INFO] [1562711982.883225457]: num_filters: 0
> [ INFO] [1562711982.883606236]: SELECTED BASE:Pose, 0
> [ INFO] [1562711982.885321016]: RealSense Node Is Up!
> [ INFO] [1562711982.885711899]: Setting Dynamic reconfig parameters.
> [ WARN] [1562711982.941155027]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
> [ INFO] [1562711983.011930398]: Done Setting Dynamic reconfig parameters.
> 22:46:45.984 [10424] [E] Device-F490: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
> 22:46:46.199 [10221] [E] Device-F490: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
> 22:46:46.199 [10221] [E] Device-F490: Entered state [ERROR_STATE]
>

echoGee, 2019-07-10:
@abernste @belkinirena : Any idea about this ?

echoGee, 2019-07-15:
Any chat forums discussing this repo development ?

doronhi, 2019-07-16:
Is the device working properly for long periods of time with realsense-viewer? rs-data-collect?
I am trying to figure out if this is a ROS wrapper problem, librealsense, usb port or device.

echoGee, 2019-07-31:
@doronhi it is working with realsense-viewer in a windows PC.

doronhi, 2019-07-31:
Can you try on the Raspberry PI? It just seems more likely to me that that is the source of the problem.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@echoGee Did you have chance to try what doronhi suggested? Any update for this issue? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

@echoGee Could you please update? Thanks!

echoGee, 2019-11-07:
Sorry, I'm offsite now and dont have access to the camera till end of Nov.

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@echoGee Thanks for the feedback! Will check with you later.

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@echoGee  Did you get chance to try and update? Thanks!

RealSenseCustomerSupport, 2020-01-15:
----------------------------------------------

@echoGee Sorry that we didn't get response for weeks. Will close it at this point. Please feel free to create new one if still have issues. Thanks!

closed 2020-01-15 by RealSenseCustomerSupport