realsenseai/realsense-ros #849: T265 drifting

reyes73a, 2019-07-11:
Hi,

I have librealsense running on a **Raspberry Pi 3B+** and a **Jetson Tx1**. I can run the rs_pose demo for T265 fine in both machines. After going through the librealsense-ros installation I run the provided examples: **roslaunch realsense2_camera rs_t265.launch** and **roslaunch realsense2_camera demo_t265.launch**. When rviz starts the pose looks good, but a couple seconds into having started the t265 pose starts drifting away in some random direction without stopping. This issue is very consistent and happens 90% of the time. I have tried:

**Raspberry Pi 3B+ running Ubuntu Mate 18 with ROS-Melodic.
Jetson TX1 running Ubuntu 16.04.6 LTS with ROS Kinetic.** 

In one case I am running **realsense2 2.23.0** and in another **2.24.0**. The librealsense-ros version was installed using **git clone https://github.com/IntelRealSense/realsense-ros.git** from (**https://github.com/IntelRealSense/realsense-ros** instructions). In both cases I get the exact same drift behavior with the T265. However, If I run the rs_pose example directly from the realsense sdk it works just fine. Therefore this is definitely a ROS wrapper issue... I have also tried on an intel i7 running ubuntu 16.04 LTS and ros kinetic and it seems to work there! I have no idea what is going on. 

Any thoughts?

Thanks!

reyes73a, 2019-07-11:
UPDATE: 
I tried running the ROS node remotely through SSH such that the only usb device connected to the TX1 is the T265 and it seem to work for the TX1. However, this is still not working for the Raspberry Pi 3B+. This is very strange. It is perhaps related to USB on the Pi not being 3.0? but then why would it work for the regular SDK example and not for the ROS example? :S I am very confused!

echoGee, 2019-07-13:
I am also seeing this issue. We are running raspberry pi 3b+ with master branch for  https://github.com/IntelRealSense/realsense-ros.git . 
We checked on windows realsense viewer and the exact drift is visible

ducklabs1, 2019-07-25:
I experienced the same behavior on a TX2 with Ubuntu 18.04. 
MY FIX:
I wrote my own realsense-ros implementation based on the rs_pose example of librealsense and the realsense-ros wrapper. Now I don't see any weird drifts happening.

doronhi, 2019-09-09:
The ROS wrapper doesn't change the pose values that librealsense provides, besides their signs and direction so it's difficult to explain the situation you describe with a realsense2_camera bug.

I assume all the tests use the same T265 device.

Could you make sure that both realsense2_camera and rs_pose use the same version of librealsense SDK? 
realsense2_camera prints it's librealsense2 SDK's version when starts running.
For checking rs_pose use `ldd ./rs-pose` to check the version.

Another thing to check is the power supply for the T265 device. Could you connect it using a hub with power supply?

nikhilnagaraj, 2019-09-25:
I see the same issue on **Intel UP squared with Ubuntu 16.04 running ROS-Kinetic.**  The pose keeps drifting in a random direction and doesn't stop. I am using a USB 2.0 hub (not separately powered) .

Does this have something to do with the fact that it's connected to a USB 2.0?

doronhi, 2019-09-26:
> The ROS wrapper doesn't change the pose values that librealsense provides, besides their signs and direction so it's difficult to explain the situation you describe with a realsense2_camera bug.
>
> I [...]

Connecting to USB 2.0 shouldn't be the source of the problem.
Do you see the difference, as mentioned before, between rs_pose and realsense2_camera? If so, can you provide the information I asked for regarding the versions?

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Hi @nikhilnagaraj, @reyes73a, @echoGee

Is anyone able to provide information that doronhi is asking?

Thanks

reyes73a, 2019-10-04:
Try USB 3.0 on the board but it shouldn't make a difference. I only see a
difference  with USB 3.0 when broadcasting a lot of data like d435. I have
the exact same setup without drift issues. I have noticed that the drift
issue only happens when you run through Ros. If you run the c++ example
directly there is no drift. This makes me think that this is a Ros wrapper
or Ros code issue perhaps. You can try making your own Ros node too and I
bet the problem will not show up any more. Is as if the timestamp was
getting injecting into the pose leading to a continuous constant drift
(e.g. parsing issue?). I have not looked at the Ros code though.

RealSenseCustomerSupport, 2019-10-18:
----------------------------------------------

Thanks @reyes73a for the information.  Although it doesn't seem like it should be the difference there between what ros wrapper is providing and c++ example.  I see that @ducklabs1 had mentioned that taking the librealsense examples and making own ros implementation that it fixed the problem.  Are you able to do that and get past the issue?

reyes73a, 2019-10-21:
@RealSenseCustomerSupport  I have not tried this, but it is on my list of things to try. I know it sound stupid because it should not make a difference, however this is what I see with 100% reliability so far. C++ examples never fail, but ROS nodes will. Not sure whats different about them, but clearly something is different about adding the ROS layer which is making it fail. Sometimes it takes 20s after starting the node for the drift to begin, other times it begins right away and other times it runs fine (quite unlikely). This is a behavior I see with RPi 3B+ and Jetson TX1. I recently reinstalled on an UpBoard and it has been working fine all the time (never seen the issue).

RealSenseCustomerSupport, 2019-10-23:
----------------------------------------------

@reyes73a - Thanks for the feedback.  And it working on the UpBoard, that's also with ROS wrapper working?

ChrisMOxon, 2020-01-16:
Hi, I'd just like to add that we're getting the same issue with USB 2 as mentioned, that there seems to be an immediate error with ROS that you get a pretty constant velocity error with the camera disappearing off into the distance.  This is on a raspberry pi 4.  When using the realsense viewer on the same device you don't get the same velocity error.

What we have noticed though separate to this is regardless of whether you're using a pi or a desktop machine the performance over USB 2 is appalling.  There are errors of up to 50% with any kind of motion.  This is using the realsense viewer.  What is strange though is we can still see the video being streamed over usb 2 - is this something we should be disabling in order to get the advertised performance and if so how do we do it?

yanjundream, 2020-02-06:
> @reyes73a - Thanks for the feedback. And it working on the UpBoard, that's also with ROS wrapper working?

I encountered exact same problem now, with librealsense V2.29.0 and realsense-ros V2.2.9. The combination of software works pretty well on a laptop (Dell XPS15), however, it is drifting when coming to TX1. I also check the result from /usr/bin/rs-pose and it is not drifting at all. Any clues what should I do? Thanks!

alanypf, 2020-03-07:
Hi,
Same problem here. I have a raspberry pi 4 and a 3B+. I tested them with the same SD card running with kernel 4.19.50-v7 and the latest development branch. The weird drifting only happened on 3B+, not raspberry pi 4. Not sure whether it helps to locate the problem or not.
Thanks.

yanjundream, 2020-03-08:
> I encountered exact same problem now, with librealsense V2.29.0 and realsense-ros V2.2.9. The combination of software works pretty well on a laptop (Dell XPS15), however, it is drifting when coming [...]

After lots of testing, we found one key problem is related to USB3. The USB port is not able to transfer both images and pose correctly through realsense-ros with our setup (TX1 and Elroy Carrier board). We tried with  /usr/bin/rs-pose-and-image sample and found the frequency of images get stuck easily. If we disable the fisheye images in the rs_t265.launch file, then the pose topic becomes very stable. 

It is interesting that when we use a shorter USB3 cable (than the original blue cable), the  /usr/bin/rs-pose-and-image works well. We doubt it is because the power supply from the USB3 of Elroy Carrier board is not enough for the transmission.

yanjundream, 2020-03-08:
> Hi,
> Same problem here. I have a raspberry pi 4 and a 3B+. I tested them with the same SD card running with kernel 4.19.50-v7 and the latest development branch. The weird drifting only happened on 3B+, not raspberry pi 4. Not sure whether it helps to locate the problem or not.
> Thanks.

I checked the 3B+ has only USB 2.0 ports, so it won't get the correct transmission of both images and pose. If you disable the images in the launch file, it probably can work well.

alanypf, 2020-03-09:
@yanjundream You are right. It works well without image publishing. Thank you very much.

closed 2020-09-11 by RealSenseSupport