realsenseai/realsense-ros #851: realsense-ros publish unorganized pointcloud

atinfinity, 2019-07-12:
Current `realsense-ros` publish unorganized pointcloud.  
<https://github.com/IntelRealSense/realsense-ros/blob/2.2.7/realsense2_camera/src/base_realsense_node.cpp#L1889-L1890>

But, I think that organized pointcloud is easier for PCL to handle..

Kaju-Bubanja, 2019-07-30:
Any idea how to change this or update this? I remember using the realsense ros wrapper before and getting organized pointclouds, did they change something?

doronhi, 2019-07-30:
It used to be organized. It was changed in the last year but I can't recall why now.
I guess a flag indicating the desired format could be nice.
I am not sure if it is a trivial addition or not. 
If you would like to add this capability, you can start with checking the code responsible for constructing the point cloud [here](https://github.com/IntelRealSense/realsense-ros/blob/5ef502ebcee5508adc073d228b4ed52d07acd5f7/realsense2_camera/src/base_realsense_node.cpp#L1886)

atinfinity, 2019-07-30:
Thank you for your comment!  
I think that you can get organized pointcloud via `depth_image_proc` nodelet.

* http://wiki.ros.org/depth_image_proc#depth_image_proc.2BAC8-point_cloud_xyz
* http://wiki.ros.org/depth_image_proc#depth_image_proc.2BAC8-point_cloud_xyzrgb

Because, [rs_rgbd.launch](https://github.com/IntelRealSense/realsense-ros/blob/2.2.7/realsense2_camera/launch/rs_rgbd.launch) uses `depth_image_proc` nodelet.

The output of these nodelet is organized pointcloud.  
https://github.com/mjgarcia/image_pipeline/blob/master/depth_image_proc/src/nodelets/point_cloud_xyz.cpp#L72

atinfinity, 2019-08-02:
I identified a changed commit. https://github.com/IntelRealSense/realsense-ros/commit/043be569e1334ba447fe85bca1a024db37085ed6

SoraDevin, 2020-01-06:
Has this been fixed/updated yet? It appears that the relevant code change from the commit referenced above by @atinfinity is on lines 1094-1101 before the changes (lines 1200, 1201 on the new).

atinfinity, 2020-01-08:
@SoraDevin It's easy to change. But, I think we need to verify the impact(peformance, relevance to other features) of this change.

doronhi, 2020-03-31:
A while ago the method of sending the smaller buffer of unorganized point cloud was chosen. As I mentioned, I think it will be beneficial to have an option to choose between the 2 methods, organized and unorganized. This option will be tackled according to priorities. a PR will be welcomed.

atinfinity, 2020-04-01:
@doronhi Thank you for your reply. I'm busy now. But, I'll send PR. So, can I ask your advice later?

RealSenseSupport, 2020-07-02:
@atinfinity Did you have a chance to submit a PR?

atinfinity, 2020-07-02:
@RealSenseSupport Sorry. Now, I don't have any time to implement.

el-stinjo, 2020-11-19:
Hi guys, 

Are there any updates on the matter? I will take a look at the commit and see if I can hack in the organized pc's.

doronhi, 2020-11-22:
No updates yet. As @atinfinity mentions, you can use the node's rs_rgbd.launch in the meanwhile to have an ordered pointcloud.

theolitl, 2020-11-25:
I am using `rs_rgbd.launch` for obtaining an organized point-cloud. 
I want high resolution depth (1280-720) but only need 640-480 for the color resolution.
A problem with this is that I am bound to the color sensor field of view for the point-cloud because `depth_registered/points` is aligned with the color frame. Is there a way I can use `rs_rgbd.launch` and gain an organized point-cloud with the FOV of the depth resolution?
So far I am using the `depth_image_proc` nodelet wich works fine, but I was wondering if it would be possible only using `rs_rgbd.launch`?

atinfinity, 2020-11-25:
@theolitl [rs_rgbd.launch](https://github.com/IntelRealSense/realsense-ros/blob/2.2.7/realsense2_camera/launch/rs_rgbd.launch) launches `depth_image_proc/point_cloud_xyzrgb` to obtain XYZRGB point cloud.
If you need to obtain XYZ point cloud, you can use `depth_image_proc/point_cloud_xyz`.
<http://wiki.ros.org/depth_image_proc>

doronhi, 2021-01-17:
Add #1627 to enable publishing pointcloud in an ordered format. I didn't test if it has benefits over using depth_image_proc or rgbd_launch

RealSenseSupport, 2021-01-20:
@atinfinity I believe this should resolve your issue so I will close this in 7 days unless you have further questions.

RealSenseCustomerSupport, 2021-01-23:
----------------------------------------------

@atinfinity, thanks! If you have any new issues please feel free to open a new ticket.

closed 2021-01-23 by RealSenseCustomerSupport

swiz23, 2021-01-25:
Just out of curiosity...

When you say organized point clouds are easier to handle in PCL, does that mean the PCL algorithms run quicker (are more efficient)? Or is it that some PCL functions expect an organized pointcloud as input, so you can't use them unless you have an organized cloud?

Also, this is probably a stupid question, but when setting the new `ordered_pc` launch file argument to true - is the processing done to get the organized pointcloud happening in the camera firmware (and all the realsense nodelet is doing is setting some flag to let the firmware know to output an organized point cloud) or on the computer side in the realsense node?

doronhi, 2021-01-26:
@swiz23 , I'll refer to the second part of your question - The whole pointcloud building is done on the computer side.
The firmware produces the depth image stream and the color image stream.
Synchronizing the streams and calculating the 3D position of every depth pixel and the matching color pixel is done in the librealsense2 library.
Converting that information into a poincloud format, either ordered or unordered, is done in the realsense2_camera wrapper.

arif0342, 2021-04-17:
> @theolitl [rs_rgbd.launch](https://github.com/IntelRealSense/realsense-ros/blob/2.2.7/realsense2_camera/launch/rs_rgbd.launch) launches `depth_image_proc/point_cloud_xyzrgb` to obtain XYZRGB point cloud.
> If you need to obtain XYZ point cloud, you can use `depth_image_proc/point_cloud_xyz`.
> Plese refer to http://wiki.ros.org/depth_image_proc.

> @theolitl [rs_rgbd.launch](https://github.com/IntelRealSense/realsense-ros/blob/2.2.7/realsense2_camera/launch/rs_rgbd.launch) launches `depth_image_proc/point_cloud_xyzrgb` to obtain XYZRGB point cloud.
> If you need to obtain XYZ point cloud, you can use `depth_image_proc/point_cloud_xyz`.
> Plese refer to http://wiki.ros.org/depth_image_proc.

Respected Member,
I am using D415 with Ubuntu 20.04 and ros-noetic. I have to detect the volume of sand from 3d point cloud so I have clone the pkg depth_image_proc but I doesn't understand how to use this pkg. is I have to write any launch file or something to get the organized 3d point cloud or what commands should I have to write in terminal in order to access this pkg? Actually I am really much new to ROS so I am very much confused to detect the sand through 3d point cloud. I have attached an image for your reference
[image]
. If you people can help me I will be very much thankful to you.

tugbakara, 2022-09-16:
> When you say organized point clouds are easier to handle in PCL, does that mean the PCL algorithms run quicker (are more efficient)? Or is it that some PCL functions expect an organized pointcloud as input, so you can't use them unless you have an organized cloud?

Actually, I am wondering that part too.

SoraDevin, 2022-09-20:
> Actually, I am wondering that part too.

I believe it is the latter

tugbakara, 2022-09-20:
I am looking for exact answer actually.

SoraDevin, 2022-09-20:
@tugbakara https://pointclouds.org/documentation/

atinfinity, 2022-09-20:
@tugbakara For example, `pcl::filters::Pyramid` could not handle unorganized point cloud.  
<https://github.com/PointCloudLibrary/pcl/blob/pcl-1.12.1/filters/include/pcl/filters/impl/pyramid.hpp#L54-L58>

And, some algorithms switch processes depending on the return value of the `isOrganized()`. Because, organized point cloud has easier access to neighboring point clouds.
<https://github.com/PointCloudLibrary/pcl/blob/pcl-1.12.1/filters/include/pcl/filters/impl/bilateral.hpp#L87-L91>

tugbakara, 2022-09-20:
Thanks @atinfinity , I am gonna try to implement unordered to ordered conversion from point cloud aggregator :) Have a nice day! :)

Pran-Seven, 2022-10-07:
Hey @tugbakara, any ideas on how to convert unordered to ordered pointcloud? Thanks in advance!

tugbakara, 2022-10-10:
> Hey @tugbakara, any ideas on how to convert unordered to ordered pointcloud? Thanks in advance!

Hi, unordered pointcloud works well Nvidia Jetson that's why I changed the SDK version of the Realsense and I stopped to use nodelets that uses CPU full level. And in codes not all types can be convertable to ordered, unoredered ones have to be same height and width datas.