realsenseai/realsense-ros #852: can't get data of infra2 with d435i

wqwanghanzhang, 2019-07-12:
I have installed the newest version ros wrapper today,and after i run "roslaunch realsense2_camera rs_camera.launch"the information from the screen tells "[ WARN] [1562935992.000370501]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 640, Height: 480, FPS: 30",and I can not get the data of infra2,but other data like infra2 is good,so how can i handle this problem? is there any connection with the firmware version?because the firmware version in my d435i is 5.10.13,maybe i should upgrade to the newest version?or the camera is broken?

[https://github.com/wqwanghanzhang/my_picture/blob/master/Screenshot%20from%202019-07-12%2020-53-25.png](url)

doronhi, 2019-07-14:
Your device was connected as a USB2, not USB3. Either you plugged it into a USB2 socket or it was accidentally identified as a USB2 socket. If the later is the case, try unplugging and plugging it again.
If still not working, make sure you followed the drivers installation procedure correctly (librealsense2-dkms).
This problem has nothing to do with your firmware.

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@wqwanghanzhang Did you get through the issue with the suggestion from doronhi? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@wqwanghanzhang Sorry that we didn't get update from you for weeks. Will close the ticket at this point. Please create another one if you still have questions/issues. Thanks!

closed 2019-12-02 by RealSenseCustomerSupport