realsenseai/realsense-ros #854: error on compiling on AWS RoboMaker Cloud9

gbr1, 2019-07-17:
Hi!
I'm trying to install latest release on Cloud9 (ROS kinetic) but a dependency error occurs: `ros-kinetic-librealsense2`.

I tried to install using: `sudo apt-get install ros-kinetic-librealsense2` but package could not be found.

I solved using an old tag release (2.2.5).
Do you know how could I get 2.2.7 working?

Thank you in advance!

doronhi, 2019-07-17:
Just remove the dependency from the package.xml file and install librealsense2 according to the instructions on librealsense2 github.
realsense2_camera naturally needs librealsense2 installed but as you say, the package itself is not yet available as a debian in ros distribution.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@gbr1 Did you get it through with doronhi's suggestion? Looking forward to your update. Thanks!

gbr1, 2019-11-01:
Yes, it works!
You can find my guide here (UP Squared AWS RoboMaker development kit):
https://github.com/gbr1/upbot_ros/wiki

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@gbr1 Glad to hear this resolved and thanks very much for the sharing! Close this accordingly.

closed 2019-11-07 by RealSenseCustomerSupport

doronhi, 2019-11-10:
> Yes, it works!
> You can find my guide here (UP Squared AWS RoboMaker development kit):
> https://github.com/gbr1/upbot_ros/wiki

Hi,
Just wondered, why in your guide you clone and compile ddynamic-reconfigure ( in section: 7. Test Realsense with ROS ) instead of installing the debian package...?

gbr1, 2019-11-10:
@doronhi probably at that time it was not available as debian package, yet.
If you have included in package dependencies, it should automatically install using rosdep, so probably I can eliminate that line.

gbr1, 2019-11-10:
@doronhi yes it works ;)
I have just updated my guide

doronhi, 2019-11-10:
Excellent. Thanks :)