realsenseai/realsense-ros #855: How to edit camera calibration parameter to /camera/color/image_raw topic

npd26, 2019-07-17:
Hello,

I have camera calibration matrix using this tutorial, [http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration](url) 
how can I add this data to /camera/color/camera_info topic manually?

thanks.

cichard, 2019-09-17:
Images are rectified internally by the firmware by using parameters you can retrieve using `rs-enumerate-devices -c`. Use this command in a prompt and look for your streaming configuration.
So many solutions offer to you depending on how you get the “new parameters”. 
* If you deal with infrared and you use the Y16 format (rectified format) as it is the case in the Realsense ros wrapper to get the images you have to apply them in the code directly [here](https://github.com/IntelRealSense/realsense-ros/blob/da4bb5dc3e29c25be9e48e4d4a66b2acdb7b1150/realsense2_camera/src/base_realsense_node.cpp#L1711) 
⚠️  camera info retrieve the values from the firmware, so you have to take them into account when changing those values 
* If you deal with RGB, RGB are also rectified internally (on board), I do no know if you can access the unrectified stream passing any option. 

What I will recommend is go and visit that [webpage](https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide) 
and look at this [document](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400%20_Custom_Calib_Paper.pdf)

jpapon, 2019-09-21:
Can I ask why the driver calls the color images image_raw if they are already rectified?

doronhi, 2019-09-25:
You cannot change the values in /camera/color/camera_info directly. 
You can write your own node that will publish different values or modify the code of realsense2_camera to publish your own values. 
The calibration tool described in the link provided by @jpapon can calculate and set new values to the device but you Intel does not provide a tool to set your own calibration values to the device's firmware.

Rectifying relates to the projection of infra2 to infra1 so that the transformation between them will consist of translation (i.e. baseline) only.
The images in topics /camera/infra1/image_rect_raw and /camera/infra2/image_rect_raw had undergone this process.
The images in topic /camera/color/image_raw are not rectified and you can see, using the `rs-enumerate-devices -c` command that the transformation from color to infra1, for example, consist of more then just translation.

RealSenseCustomerSupport, 2019-11-08:
----------------------------------------------

Hi @npd26,

Any update? Did you also refer to the custom calibration guide to have your own calibration?
Note: unrectified frames are exposed in Y16 format for calibration.

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

If no further question, will close this.

closed 2019-12-05 by RealSenseCustomerSupport