realsenseai/realsense-ros #857: depth measurement

GESZ24, 2019-07-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Firmware Version                   | Open RealSense Viewer  | 
| Operating System & Version |   Win (10) / Linux (Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  ( 4.18.0-15)                                         | 
| Platform                                 | PC  |
| SDK Version                           |      2.23.0.909                   | 
| Language                              |  Python                          | 
| Segment	            		|  others                          | 

### Issue Description
I´m working with depth camera D435 and I´m trying to measure short distances (mm) but i can´t change the units, I´m using the camera to mesuare the depth wheel but sometimes the depth is detected but the change that show the viewer is minimum 1 cm.

[image]
[image]

MartyG-RealSense, 2019-07-18:
It looks as though you are using the Depth Quality Tool to take a depth measurement.   The appropriate tool to use for that application would be the RealSense Viewer tool, since the Depth Quality Tool is meant to be used to assess the quality of the data being processed by the camera.

The Depth Units control in the RealSense Viewer allows you to use a slider to change the depth units from the default of 0.001 to a minimum of 0.0001 for improved close-range depth sensing accuracy.
https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality

RealSenseCustomerSupport, 2019-10-30:
----------------------------------------------

Closing this unless there is any other question. Thank you.

closed 2019-10-30 by RealSenseCustomerSupport