realsenseai/realsense-ros #860: D345i lower  frame rate

phil333, 2019-07-19:
I would like to put the realsense 435i frame rate down to 5fps, as I have a rover that moves fairly slowly, and 15fps, is more than i can process in realtime. It looks like the ros-wrapper is refusing to go lower than that?

MartyG-RealSense, 2019-07-19:
The 400 Series cameras support a minimum of 6 FPS.  So you should be able to go below 15 FPS to 6 FPS.

I would not personally recommend 6 FPS mode, as it has caused some RealSense users to experience errors such as frames not arriving in a timely manner.  They found that the problems disappeared when they used a minimum of 15 FPS.

However, if you really do need 6 FPS, an approach taken in ROS by one user was to control the FPS by deactivating auto-exposure and manually controlling the exposure with rqt_reconfigure.

https://devtalk.nvidia.com/default/topic/1036709/jetson-tx2/ros-realsense-d435-on-jetson-tx2-only-15fps-fhd-availlable/post/5267670/#5267670

phil333, 2019-07-22:
@MartyG-RealSense thank you for the tip, i will check this out

phil333, 2019-07-23:
@MartyG-RealSense 6fps is somehow possible, but it is not very stable, and keeps throwing error messages. Anyway, i m looking for 5fps because i m benchmarking it against other sensors, so ideally they all have the same rate. Does not really make sense to me that a low frame rate is not allowed

MartyG-RealSense, 2019-07-23:
The more FPS or resolution modes that are added, the more test time that would be required for each new SDK release, which could be awkward for a rapid-development cycle such as the RealSense SDK's, where each release comes on average one month after the previous one.  

For example, a number of users requested a 24 FPS mode because it would make the camera more useful for movie work, but the RealSense developers decided against adding a new FPS mode to support that.  

I would not exclude the possibility that further modes could be added through technology innovations.  Going a bit off-topic, some videogame development companies rapidly test products that undergo regular updates by having an AI check them for bugs after changes are made to the code, like the example of the videogame 'Sea of Thieves'.

https://www.youtube.com/watch?v=Bu4YV4be6IE

phil333, 2019-07-23:
Thx for the clarification. For our robotic systems, the frame rate doesn't make sense, as we can't process the images in real-time, so it just clogs up our rosbags. We re only interested in rgbd + imu.
If you can point me towards where the definitive frame rate limit is set in the SDK, i can also start experimenting on my own.

MartyG-RealSense, 2019-07-23:
The FPS modes are likely defined in the camera's firmware driver (one of the Intel guys can confirm if this is so - I'm not an Intel employee) and Intel do not open source the firmware.  If it is in the firmware then there is likely nothing you can do directly and would have to do some indirect measure to add 1 FPS worth of artificial drag to a 6 FPS stream. 

I wonder if you could purposely introduce a little latency by setting the Frame Queue value to be sub-optimal.

https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

MartyG is correct, the frame rate is set by firmware and currently the closest frame rate to 5FPS is the 6FPS.

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

phil333, 2019-10-31:
Thank you for the information. I'm the end we modified the Ros wrapper to run at a higher frame rate, but to only publish 1 out of 3 images. Works fine for our needs.

closed 2019-10-31 by phil333

lostmsu, 2020-02-08:
@phil333 can you share the modification that you made?

phil333, 2020-02-12:
> @phil333 can you share the modification that you made?

Hey, i dont have access to my code, but i can tell you want to do. So my modification consists in skipping the publication of 2/3 images. For this you have to go to the following method inside the base_realsense_node.cpp:
`void BaseRealSenseNode::frame_callback(rs2::frame frame)`
There you add a simple if else clause before the 'try{' line, and go to return if you want to skip the image publication this time. You can do this based on a class variable inside the .h which you already init to 0