realsenseai/realsense-ros #861: How to get RGBD Depth Image Stream?

amussell, 2019-07-19:
I am trying to get an RGBD image stream from the realsense. I can see in the docs where there is an RGBD point cloud stream, but this is not the format I need the data in. I need the image color and depth data in a 4 channel image array.

Ultimately what I want to do is calculate a mask of an object in an image using the color image, then use that mask to mask the depth image and get a point cloud for the object. I need to be able to do 1 of 2 things: 1. Get an RGBD image stream from the node, or project the RGBD point cloud on to an RGBD image.

How can I get an RGBD image?

doronhi, 2019-08-06:
I think the more straight forward approach would be to subscribe the color and aligned_depth_to_color topics.
run the node with the following command:
```
roslaunch realsense2_camera rs_camera.launch align_depth:=true
```
to have the following topics:
`/camera/aligned_depth_to_color/image_raw`
`/camera/color/image_raw`

The images are aligned and use can mask pixels from one image to the other.

RealSenseCustomerSupport, 2020-01-17:
----------------------------------------------

Hi @Levi568,

Please make sure that you can read color images from /camera/color/image_raw.
Having color images from that topic and aligned (depth to color) images from
/camera/aligned_depth_to_color/image_raw you can easily get RGBD image stream.  Let us know whether you are able to obtain the stream you want and whether or not you need any additional help with this question.

Thank you!

closed 2020-01-23 by RealSenseCustomerSupport

ranjitkathiriya, 2020-11-19:
Hi @doronhi,

Actually, I am getting RGB-D from the stream but it is in black color can you please help me with that here I am posting my code and Image of the output.

`#!/usr/bin/env python3
import rospy
from sensor_msgs.msg import Image
from cv_bridge import CvBridge 
import cv2
import numpy as np
 
def callback(data):
  br = CvBridge()
  rospy.loginfo("receiving video frame")
  current_frame = br.imgmsg_to_cv2(data)
  cv_image_array = np.array(current_frame, dtype = np.dtype('f8'))
  cv_image_norm = cv2.normalize(cv_image_array, cv_image_array, 0, 1, cv2.NORM_MINMAX)
  # cv2.imshow("camera", cv2.cvtColor(current_frame,cv2.COLOR_BGR2RGB))
  cv2.imshow("camera", cv_image_norm)
  print(cv_image_norm.shape)
  cv2.waitKey(1)
      
def receive_message():
  rospy.init_node('video_sub_py', anonymous=True)
  rospy.Subscriber('/camera/aligned_depth_to_color/image_raw', Image, callback)
  rospy.spin()
  cv2.destroyAllWindows()
  
if __name__ == '__main__':
  receive_message()`

[image]

Thank you so much for the help

GauravKakade, 2021-02-04:
@ranjitkathiriya you can apply colormap to your stream 
for eg. depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(cv_image_array, alpha=0.03), cv2.COLORMAP_JET)

ranjitkathiriya, 2021-02-04:
Even by setting filters:=colorizer, you can get the depth color stream.

`roslaunch realsense2_camera rs_rgbd.launch filters:=colorizer align_depth:=true`