realsenseai/realsense-ros #863: DO not need image from T265 ROS node

deeprun, 2019-07-20:
Hi, I installed T265 on my robotcar. The car runs on Nvidia Jetson TX2 computer, which has only one USB port. So I used a USB hub to connect T265, IMU, wireless mouse/keyboard, etc to the TX2. I noticed that once I inserted T265 USB into the USB hub, my mouse and keyboard became much slow in response. Why? Is that because images are transferred from T265 to the car through the hub? Actually, I do not need images from T265, I just need the position information. Can I prevent the images being send back to the car?

MrG4dget, 2019-07-21:
Well not sure about the USB issue, it might be a bandwidth issue. you can check if the system runs better if you disable image streaming. 

In the t265 launch file (line 15 & line 16) set both cameras to false. This should prevent the node from publishing the images.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_t265.launch

```
 <arg name="enable_fisheye1"     default="false"/>
 <arg name="enable_fisheye2"     default="false"/>
```

hope it helps!

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Hi @deeprun

I'm going to go ahead and close this issue ticket since MrG4dget provided sufficient response to your question.

Thanks

closed 2019-10-04 by RealSenseCustomerSupport