realsenseai/realsense-ros #864: Change T265 ros topic name

deeprun, 2019-07-20:
If I ran T265 launch file rs_t265.launch, all topic are published with a /camera prefix. How can I change it to something else?

hoangthien94, 2019-07-22:
The prefix is determined by the argument `camera`, located in the launch file `rs_t265.launch`. 

You can change it [here](https://github.com/IntelRealSense/realsense-ros/blob/31fa75ec835650d319f2d5abd3f7d0ab66fd37f4/realsense2_camera/launch/rs_t265.launch#L10)  or [pass it into the command line](http://wiki.ros.org/roslaunch/XML/arg#Passing_an_argument_via_the_command-line).

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Hi @deeprun, I'm going to go ahead and close this issue out since hoangthien94 has provided clear direction to answer your question.

Thanks

closed 2019-10-04 by RealSenseCustomerSupport