realsenseai/realsense-ros #865: doesn't realsense-ros support Ubuntu 14.04 ROS Indigo enviroment?

suho0515, 2019-07-23:
Hello, I'm trying to work with my Intel realsense D435. 
in ROS wiki (http://wiki.ros.org/realsense2_camera), there is page for indigo enviroment.
and in Github page (https://github.com/IntelRealSense/realsense-ros), it's mentioned that [The following instructions support ROS Indigo, on Ubuntu 14.04, and ROS Kinetic, on Ubuntu 16.04.]
. 
but [latest Intel® RealSense™ SDK 2.0] need at least Ubuntu 16.04 OS. 
does it mean we can not operate camera in Ubuntu 14.04 OS? and also ROS indigo?

Look forward hearing from you.
Regards.

doronhi, 2019-07-23:
@suho0515 ,
This is an error in documentation. I am sorry it misled you. I know from previous questions that the version does not compile as is on indigo release, see #748 .
The version is officially supported in Ubuntu 16.04 and apparently also on Ubuntu 18.04.
I apologize again and I will fix the documentation error.

ravijo, 2019-09-25:
@suho0515: Just for the information, if you really want to use ROS Indigo, you can use version 2.2.1 of `realsense-ros`.

Please see [this link](https://github.com/IntelRealSense/realsense-ros/issues/748#issuecomment-534938772) for more info.

Cheers

RealSenseCustomerSupport, 2019-11-08:
----------------------------------------------

Hi @suho0515, Do you need further help on this?
Thanks,

suho0515, 2019-11-08:
I had a enough information for this issue, thanks.
I'd close this issue :)

closed 2019-11-08 by suho0515