realsenseai/realsense-ros #866: Error syncing TM2 and Host clocks && ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0

joekeo, 2019-07-23:
Hello, 

I keep trying to debug this issue with the T265.
The T265 starts fine, but sometime after (can be 1 minute or an hour) the sensor starts sending this error:

```
08:45:27.942 [24746] [E] Device-6C40: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
08:45:27.942 [24746] [E] Device-6C40: Error syncing TM2 and Host clocks
08:45:28.446 [24746] [E] Device-6C40: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
08:45:28.446 [24746] [E] Device-6C40: Error syncing TM2 and Host clocks
08:45:28.950 [24746] [E] Device-6C40: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
08:45:28.950 [24746] [E] Device-6C40: Error syncing TM2 and Host clocks
08:45:29.454 [24746] [E] Device-6C40: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
```
While this error is happening, the T265 still works as intended, but after a while of sending the error, the realsense tends to crash.

If the node is killed before it crashes, it can be started again without problems, and the error is not present at the startup (keeps behaving the same, after a while it appears).

But if the T256 crashes, it can only be started again after a reboot.

PD: this error is also sometimes displayed:
```
09:20:20.480 [25695] [E] Device-6C40: Unsupported message (0x6F60) on stream endpoint thread function
09:21:16.247 [25695] [E] Device-6C40: Unsupported message (0x8081) on stream endpoint thread function
```

SteveGrehl, 2019-08-01:
I've the same issue on a  `Ubuntu 16.04.5 LTS` using `Realsense ROS v2.2.7` and `LibRealSense v2.24.0` during start it states the following warning:

>  01/08 10:25:56,051 WARNING [139981005596416] (types.cpp:48) hardware_reset is not implemented for this device!
[ WARN] [1564647956.052707066]: An exception has been thrown: hardware_reset is not implemented for this device!
> ...
> [ WARN] [1564647956.314680764]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)

I assume it is a clock synchronization issue. Is there a way to set the time on the T265?

SteveGrehl, 2019-08-07:
Since there was an update to `LibRealSense v2.25.0` this specific error was fixed for me. However, I get now:

```
[ INFO] [1565182277.811417739]: Initializing nodelet with 8 worker threads.
[ INFO] [1565182277.844241152]: RealSense ROS v2.2.7
[ INFO] [1565182277.844284595]: Running with LibRealSense v2.25.0
14:51:17.869 [6922] [E] Device-F600: ERROR: Bulk transfer message 0x25 (DEV_SET_LOW_POWER_MODE) request to device got LIBUSB_ERROR_IO. Bytes requested 8, bytes transferred 0
14:51:17.872 [6922] [E] Device-F600: ERROR: Bulk transfer message 0x1 (DEV_GET_DEVICE_INFO) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
14:51:17.872 [6922] [E] Device-F600: USB Error (0x6)
14:51:17.872 [6922] [E] Device-F600: Error: Failed to get device info (0x6)
14:51:17.876 [6922] [E] Device-F600: ERROR: Bulk transfer message 0x1 (DEV_GET_DEVICE_INFO) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
14:51:17.876 [6922] [E] Device-F600: USB Error (0x6)
14:51:17.876 [6922] [E] Device-F600: Error: Get Device Info error (0x6)
14:51:17.876 [6922] [E] Device-F600: Entered state [ERROR_STATE]
14:51:18.188 [6919] [E] Device-F600: Error: Got Temperature for 3043476371 sensors, exceeding max sensors (3)
 07/08 14:51:18,187 ERROR [140443938952960] (types.h:243) Failed to query T2xx tracking temperature option
 07/08 14:51:18,188 WARNING [140443938952960] (rs.cpp:281) null pointer passed for argument "device"
terminate called after throwing an instance of 'rs2::error'
  what():  Failed to query T2xx tracking temperature option
```
But this may be a different issue. I will first look into it.

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Hi @joekeo

Is the original issue still happening with the latest librealsense package?

Thanks

joekeo, 2019-10-07:
Hello,
Currently we replaced the Realsense in Two of our robots for other sources of odometry. But I will mount it back at some point this week and let you know if the error still happens with the latest software

SteveGrehl, 2019-10-07:
Just tested it with 
```
[ INFO] [1570459948.159862645]: RealSense ROS v2.2.8
[ INFO] [1570459948.159909963]: Running with LibRealSense v2.29.0
```
and still getting:

```
16:54:41.931 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:42.435 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:42.435 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:42.939 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:42.939 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:42.946 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x18 (DEV_GET_TEMPERATURE) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:42.946 [7297] [E] Device-1250: USB Error (0x6)
 07/10 16:54:42,945 ERROR [139854067267328] (types.h:304) Failed to query T2xx tracking temperature option
16:54:42.949 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x18 (DEV_GET_TEMPERATURE) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:42.949 [7297] [E] Device-1250: USB Error (0x6)
 07/10 16:54:42,949 ERROR [139854067267328] (types.h:304) Failed to query T2xx tracking temperature option
16:54:43.443 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:43.443 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:43.947 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:43.947 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:44.452 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:44.452 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:44.955 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:44.956 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:45.460 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:45.460 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:45.964 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:45.964 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:46.467 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:46.467 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:46.971 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:46.971 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
16:54:47.475 [7278] [E] Device-1250: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
16:54:47.475 [7278] [E] Device-1250: Error syncing TM2 and Host clocks
```

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Is there anyway you can provide a bit more details on the setup and a bit of sample code you are running when hitting this error?

Is the USB cable/connector secure?  Meaning at the T265 side, do you have the connector screwed in such that there is no movement of the connection points?  We've seen in the past that USB connection issues can cause such bulk transfer errors.  Also are you connecting with USB2 or USB3?  How about the cables, are they the cables that came with the devices or are you using other USB cables?

SteveGrehl, 2019-10-09:
TL;DR A reboot may have solved the issue.

I did some odometry experiments with our robot just yesterday and started the T265 during those, because why not. The error didn't appear once during the experiments (approx. 2,5h). However, I didn't look into the data. Since I'm using the ROS interface I can say at least there are Pose-Messages. I will look into this later this week.
Regarding your questions: 

> [...] a bit of sample code you are running when hitting this error?
here is the section from the ROS launch file:
```
<include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_sync"              value="$(arg enable_sync)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye1"          value="$(arg enable_fisheye1)"/>
      <arg name="enable_fisheye2"          value="$(arg enable_fisheye2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>

      <arg name="topic_odom_in"         value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"       value="$(arg calib_odom_file)"/>

      <arg name="publish_odom_tf" value="false" />
      <arg name="odom_frame_id" value="odom_ldr" />
<!--      <arg name="base_frame_id" value="rs_front_t265_base_link" />-->
      <arg name="pose_frame_id" value="rs_front_t265_base_link" />
    </include>
```
The arguments where copied from [here](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_t265.launch).

> Is the USB cable/connector secure?
I use the cable that comes with the camera. It is not further protected. The error was raised even when the robot was at rest. 

> Meaning at the T265 side, do you have the connector screwed in such that there is no movement of the connection points?
No, I don't.

> Also are you connecting with USB2 or USB3?
USB 3 - I tried to put the camera on a USB3-Hub, but that didn't work so I connected it directly.

> How about the cables, are they the cables that came with the devices or are you using other USB cables?
As mentioned above the cable that came with the device. But there is a good chance I want to switch the cable to a longer one. What are the restrictions/specifications here?

---

Is there a way to sync the clock of the T265 with the CPU clock?

RealSenseCustomerSupport, 2019-10-18:
----------------------------------------------

Glad to see that you're not experiencing the error any longer after reboot and during your experiments.  With ROS pose data/information/messages are going through ROS wrapper layer, which would be different then not using ROS wrapper in general.  Do let me know if you were able to look at the data.

sumilmajithia, 2019-10-23:
14:33:06.508 [4956] [E] Device-7300: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0 
14:33:06.508 [4956] [E] Device-7300: Error syncing TM2 and Host clocks 
14:33:06.568 [5829] [E] Device-7300: FW crashed - got error in stream endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0 
14:33:06.572 [5828] [E] Device-7300: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0 
14:33:06.769 [4956] [E] Device-7300: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE] 
14:33:06.770 [4956] [E] Device-7300: Entered state [ERROR_STATE] 
[ERROR] [1570566787.773939710]: The device has been disconnected! 
[ INFO] [1570566788.782789113]: Checking new devices... 
[ INFO] [1570566788.817235281]: No calib_odom_file. No input odometry accepted. 
[ INFO] [1570566788.817697821]: getParameters... 
[ INFO] [1570566788.926178338]: setupDevice... 
[ INFO] [1570566788.926221304]: JSON file is not provided 
[ INFO] [1570566788.926322392]: ROS Node Namespace: T265 
[ INFO] [1570566788.926367132]: Device Name: Intel RealSense T265 
[ INFO] [1570566788.926392165]: Device Serial No: 845412111181 
[ INFO] [1570566788.926419742]: Device FW version: 0.0.18.5715 
[ INFO] [1570566788.926444379]: Device Product ID: 0x0B37 
[ INFO] [1570566788.926468343]: Enable PointCloud: Off 
[ INFO] [1570566788.926489179]: Align Depth: Off 
[ INFO] [1570566788.926514603]: Sync Mode: Off 
[ INFO] [1570566788.926611154]: Device Sensors:  
[ INFO] [1570566788.926701064]: Tracking Module was found. 
[ INFO] [1570566788.926735762]: (Depth, 0) sensor isn't supported by current device! -- Skipping... 
[ INFO] [1570566788.926769015]: (Color, 0) sensor isn't supported by current device! -- Skipping... 
[ INFO] [1570566788.926793836]: (Infrared, 1) sensor isn't supported by current device! -- Skipping... 
[ INFO] [1570566788.926820029]: (Infrared, 2) sensor isn't supported by current device! -- Skipping... 
[ INFO] [1570566788.926847252]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping... 
[ INFO] [1570566788.927110198]: setupPublishers... 
[ INFO] [1570566788.929181806]: Expected frequency for fisheye1 = 30.00000 
[ INFO] [1570566788.963621969]: Expected frequency for fisheye2 = 30.00000 
[ INFO] [1570566788.999407245]: setupStreams... 
[ INFO] [1570566788.999945237]: fisheye1 stream is enabled - width: 848, height: 800, fps: 30 
[ INFO] [1570566789.000391575]: fisheye2 stream is enabled - width: 848, height: 800, fps: 30 
[ INFO] [1570566789.000551225]: insert Fisheye to Tracking Module 
[ INFO] [1570566789.000606985]: insert Fisheye to Tracking Module 
[ INFO] [1570566789.000647725]: insert Gyro to Tracking Module 
[ INFO] [1570566789.000684860]: insert Accel to Tracking Module 
[ INFO] [1570566789.000725773]: insert Pose to Tracking Module 
[ INFO] [1570566789.027046080]: num_filters: 0 
[ INFO] [1570566789.027112362]: SELECTED BASE:Pose, 0 
[ INFO] [1570566789.027425072]: RealSense Node Is Up! 
[ INFO] [1570566789.027461132]: Setting Dynamic reconfig parameters. 
[ INFO] [1570566789.030988346]: Done Setting Dynamic reconfig parameters. 
[ WARN] [1570566789.031454585]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME) 

I am getting a similar problem here but the T265 resets on its own but it's too late for the robot since it has lost its tracking robot. I see a time difference of about 4 seconds before it starts publishing odometry data again.

yputra, 2019-11-04:
@SteveGrehl do you try your robot again with the T265? I have exactly the same issue and it does not go away even if I reboot everything.

SteveGrehl, 2019-11-14:
@yputra  I did a further experiment, to investigate the topic. I disconnected the camera from the robot and setup an isolated workspace on my laptop. I run the camera with the same cable over night without any issues. When I came back in the morning the camera was still running and an odometry was published.
Since I use the same software setup on my robot and on my laptop I assume the error is not related to this repository and the issue should be closed.
My guess is that the USB bus of the host system does not supply enough power and that this is the reason for the malfunction. One solution could be to use an active USB hub. The exact specifications of the USB interface of the T265 would help here (@RealSenseCustomerSupport).

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

The T265 uses the standard USB interface that's publicly available.

Given SteveGrehl findings I agree with his conclusion.

Will you be needing any further assistance with this before we close?

SteveGrehl, 2019-12-10:
The computer to which I connected the camera had a hardware defect at the USB interface, which was the cause for the error message. @RealSenseCustomerSupport the issue can be closed, thanks for the help.

closed 2020-02-20 by RealSenseCustomerSupport