realsenseai/realsense-ros #868: about the IMU of t265

RJJxp, 2019-07-24:
01. Can I get the correct pose data without the imu data input? How if possible?
02. What does param "unite_imu_method" actually mean with value "none", "copy", "linear_interpolation"? Does it mean the way using imu data to calculate pose data? Or just process the imu raw data and publish them?

```c++
void BaseRealSenseNode::imu_callback_sync(rs2::frame frame, imu_sync_method sync_method);
void BaseRealSenseNode::imu_callback(rs2::frame frame);
```

RJJxp, 2019-07-24:
I read the issue #558 , I am sorry I didn't get it :-P

doronhi, 2019-07-24:
1. The pose message includes velocities. That is the format of the message. This is not the IMU input though. The velocities published are based on the SLAM algorithm, i.e. include information from the images.

2. unite_imu_method param relates to the raw IMU data. Originally it comes in 2 separate, un-synchronized streams. Many applications prefer having the Gyro data and the accelerometer data to come synchronized, in the same message.
The two methods are available for this synchronization, "copy" and "linear_interpolation", are described in the README.md file (although, admittedly, the tabs should be corrected..).
Regarding your question, they are not used for pose calculation or any other internal use. It is only a convenient way to publish the data.

RJJxp, 2019-07-25:
@doronhi Thank you so much for replying!!!

According to your answer to my first question, the pose data is produced by the V-SLAM algorithms which depends on the imu data input and image data input of the 2 sensors in T265 camera. 

**I'd really like to know is there any option to force the V-SLAM algorithms to use only image data input to publish pose data?** Like a param in launch file or in the code? (well, I guess it's impossible lol) Cause My robot(vehicle) will shake constantly once started. It affects the imu sensor a lot I presume. Of course, with the input of robot's odometry data, the tracking(pose) result is much better.

Finally thanks again for your help!!!

Jwenda, 2019-08-13:
I have faced a similar issue as you @RJJxp. Our UAV odometry from T265 has drifted a lot after the take-off due to vibration from the rotors. It makes it basically impossible to use with built-in V-SLAM. Please if you know how to exclude the IMU or count with the vibration, please let me know.

Thanks,
Jan

RJJxp, 2019-08-16:
@Jwenda No problem. The project of T265 suspended in part due to short-handed, right now I am not working on this but I will let you know if progress made

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Hi @Jwenda

To answer your question, there is no way to ignore the IMU data coming into the V-SLAM Algorithm.  It is a necessary input as part of the design of T265.  For wheeled robot devices yes there is wheel odom as outlined in this issue ticket.  We do not have input for aerial devices that would be similar to wheel odom input into the T265 system.

closed 2019-10-11 by RealSenseCustomerSupport

Shivani1796, 2019-12-20:
@doronhi @RJJxp Hey, I read above messages can you tell me how to generate pose file for T265 camera ?

doronhi, 2019-12-29:
Could you rephrase your question? What do you mean by "pose file"?

Shivani1796, 2019-12-29:
@doronhi  Hi, thanks for response actually i want to get pose file ( translation & rotation matrix) from that. I also have .bag file recorded by t265 camera , if i extract it may be will get pose file but i don't have any method how to solve this issue .

doronhi, 2020-01-06:
I am not sure I understand.
Either from the .bag file, or while the realsense2_camera node is running, check the "/camera/odom/sample" topic. It contains the translation and rotation values. From them you can construct your matrices.

Shivani1796, 2020-01-06:
@doronhi  hey thanks for response,
Actually I have .bag file recorded from realsense t265 camera & i have converted the bag file to png images so i have sequence of images and for that i need pose data I hope I made you clear. So is there any method to direct convert my images to pose data or any code available.

doronhi, 2020-01-06:
The images are saved in the .bag file under '/camera/color/image_raw' topic (from memory so maybe I spelled it wrong). The pose data is in the topic I mentioned: /camera/odom/sample.
You can extract it from the bag file the same way you extracted the images.
Or maybe the issue is still not clear to me.. :)

Shivani1796, 2020-01-07:
I tried to extract bag file in csv format same as I did for images, but rs-convert tool ( https://github.com/IntelRealSense/librealsense/tree/master/tools/convert )is giving some issues to convert .bag-->.csv file.Here is the error I get after running ./rs-convert -v test -i 1.bag ( Linux OS)
[image]

I hope now I will make you clear  

> "Want to retrieve 6DOF or all data like gyro, accel, pose (rotation & translation matrix) for all the frames in a text file format.

"

doronhi, 2020-02-02:
Sorry for the long delay.
Did you record 1.bag file using realsense-viewer or using rosbag --record?
realsense-viewer and rs-convert use a different format than the rosbag application so you need to use the right one to extract the file. I am not sure it has something to do with the error you get - I don't have enough data.
In any case, you will not get pose for each image. The pose messages are not synchronized with the image messages and they come in different frame rates, although they share the same time line.