realsenseai/realsense-ros #869: Questions regarding d435i SLAM

Mightysaur, 2019-07-25:
I have recently been looking at the wiki with D435i and SLAM and have been using it. It works really well and I'm looking at ways to expand it.

My first question is: Is it possible to add multiple cameras if I modify the launch settings?
My second question: If it is possible to add multiple cameras, what model camera would be ideal as a second IMU unit may not be necessary?

Thanks for your help

MartyG-RealSense, 2019-07-25:
Intel published an article on using multiple cameras with ROS.

https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

If you are going to combine data from multiple cameras (even if not using sync cabling) I would recommend using all the same kind of camera (e.g 1 D435i and all the rest standard D435s).  This is because the D415 has a different size of field of view, different IR imager component and a different, slower shutter type.

doronhi, 2019-07-31:
The cameras themselves can work together. The issue that I foresee is that the rtabmap node, used in the wiki example, is responsible for creating the unified pointcloud. I don't know if it's interface allows for more then 1 depth/rgb topic. Maybe you could try to remap the topic of camera 2 to the topic of camera 1?

Mightysaur, 2019-07-31:
Thank you for your response. Could you please explain your last point regarding the topic? Are you saying that we ignore the output of camera 1? Or more likely what you mean is find a way such that the data from both cameras be combined into one topic?

I believe I remember reading that rtabmap currently supports up to 2 rgdb cameras and has been done with 2 Kinects so in theory should be doable with the d435 also?

doronhi, 2019-08-04:
I didn't investigate so much into the rtabmap but if it supports 2 rgbd cameras then that's the best course of action, I think.
I meant, as you said, that maybe it's possible to send both rgb images on the same topic. Maybe by keeping every camera with it's individual frame_id rtabmap will manage.
Or, as another resort, maybe there are alternatives to rtabmap.

siddhya, 2019-09-17:
What is the tracking accuracy of slam with d435i? Is it in the order of centimeters? Also, what is the drift over say 100 meters?

doronhi, 2019-09-25:
D435i does not provide SLAM algorithm. Just an IMU.
I expect the answer will depend on the SLAM implementation, environmental conditions, hardware and more, besides the D435i's own accuracy.

MartyG-RealSense, 2019-09-25:
Intel has a guide for doing SLAM with D435i by combining the IMU with other software tools.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i?language=en_US

RealSenseCustomerSupport, 2019-11-20:
----------------------------------------------

Hi @Mightysaur

Is there anything further you need on this thread?  Thanks to MartyG-RealSense and doronhi for providing some information and links.

closed 2019-12-13 by RealSenseCustomerSupport