realsenseai/realsense-ros #870: Changelog?

ubtjoe, 2019-07-25:
It would be really nice to have a changelog for each release where the following is made clear:

- Which librealsense release is this new release compatible with?
- What are the added features since the previous release?
- What has been fixed/addressed?
- What has been deprecated?

This would make my life a lot easier, and I imagine it would do the same for many developers here.

doronhi, 2019-07-31:
Do you mean a file with a collection of all the release notes given for each version?

ubtjoe, 2019-07-31:
@doronhi yes.  The release notes should include the versions of librealsense that the realsense-ros release is compatible with.

cjue, 2019-07-31:
For the tagged versions the compatible librealsense version should always be visible in the README.md and in the CMakeLists.txt file, but a separate changelog file with some additional release notes would be very helpful.

https://github.com/IntelRealSense/realsense-ros/blob/31fa75ec835650d319f2d5abd3f7d0ab66fd37f4/realsense2_camera/CMakeLists.txt#L37

https://github.com/IntelRealSense/realsense-ros/commit/31fa75ec835650d319f2d5abd3f7d0ab66fd37f4#diff-4001b635bf5f34f3c5d38c6c5568029e

doronhi, 2019-08-06:
Sorry guys, you got me a little confused.
Regarding mentioning the compatible librealsense version in the README.md file, I agree.
The comatible version is already stated in the CMakelists.txt file, as @cjue mentions.
Isn't all the rest of the information already gathered in the [releases page](https://github.com/IntelRealSense/realsense-ros/releases)?

ubtjoe, 2019-08-06:
@doronhi - YES!  The releases page has the information I am looking for, and yet there is no mention of it in the README.  IMO, creating a section in the README that points the users to the releases page would be very helpful because, as I have found, previous versions of librealsense work better on my target hardware (Jetson Xavier) and it was not immediately clear to me which realsense-ros version to use with it.

As far as @cjue comment goes: while it is true that users could root around in the CMakeLists.txt file to find the dependency, but should this step really be necessary?  Better documentation exists in the form of the releases page, so why not just make this information more transparent by referencing it in the top-level README?

doronhi, 2019-08-08:
Just released version 2.2.8 with link in the README.md to the release notes.

ubtjoe, 2019-08-08:
Looks great!

closed 2019-08-08 by ubtjoe