realsenseai/realsense-ros #871: How to set the advanced control parameters using the realsense-ros wrapper?

Kaju-Bubanja, 2019-07-30:
I opened the dynamic reconfigure with `rosrun rqt_reconfigure rqt_reconfigure`, but looking at the options in the stereo module there are no post processing options like in the realsense-viewer: 
[image]
[image]

I looked at the stereo_module with following command: `rosrun dynamic_reconfigure dynparam get /camera/stereo_module` with following output:
`{'laser_power': 183.6, 'groups': {'laser_power': 183.6, 'parent': 0, 'global_time_enabled': True, 'inter_cam_sync_mode': 0.0, 'gain': 16, 'groups': {}, 'id': 0, 'output_trigger_enabled': False, 'exposure': 33000, 'name': 'Default', 'parameters': {}, 'emitter_enabled': True, 'enable_auto_exposure': True, 'error_polling_enabled': True, 'state': True, 'visual_preset': 4, 'enable_auto_white_balance': False, 'type': '', 'frames_queue_size': 16}, 'emitter_enabled': True, 'enable_auto_exposure': True, 'frames_queue_size': 16, 'global_time_enabled': True, 'inter_cam_sync_mode': 0.0, 'visual_preset': 4, 'error_polling_enabled': True, 'gain': 16, 'exposure': 33000, 'output_trigger_enabled': False, 'enable_auto_white_balance': False}
` but there is no persistency_index parameter.
The rs_rgbd.launch file has the argument filter and I set it to "temporal" and it is mentioned in [the docs](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#temporal-filter), but I can't figure out how to actually set the values described in the docs, like the persistency index and others. I tried `rosrun dynamic_reconfigure dynparam set /camera/stereo_module persistency_index 8` but get following error: `error updating parameters: don't know parameter: persistency_index`
How do I set the advanced control parameters using the realsense-ros wrapper?

doronhi, 2019-07-31:
As you already found out, the post processing options are there only if stated with the `filters` param.
All the options of the filters should automatically be present with rqt_dynamic_reconfigure. The names may differ though.
The option listen in realsense-viewer as `persistency_mode` appears in ROS as `holes_fill`.
you can change it with the following command:
`rosrun dynamic_reconfigure dynparam set /camera/temporal holes_fill 8`

Kaju-Bubanja, 2019-08-02:
Thanks for the respons, two questions:

- Is there a list of mappings between the names? Or how are we supposed to know what the parameters are called?

- And second, then why didn't holes_fill show up for me? I launched the rs_rgbd file with the param `filters` set to `temporal` I will attach my launch file on monday for the exact wording, but I'm very sure it was set to temporal, is it maybe the wrong filter name?

Kaju-Bubanja, 2019-08-05:
`rosrund dynamic_reconfigure dynparam list` Does not show that there is a `/camera/temporal` param. I tried the above command and it just hangs and waits. Also the param is not in the rqt_reconfigure gui. So the question remains, how does one set these parameters?

Kaju-Bubanja, 2019-08-05:
Haha, I found the error and you actually fixed it in the last commit #5ef502ebcee5508adc073d228b4ed52d07acd5f7. The question for me now is why the README.md tells me to check out the second last commit and not the last? Could this line `git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`` be removed from the installation instruction? Because why should people not use the latest commit?

doronhi, 2019-08-06:
The installation instructions refer to the latest release.
The last commit often is ahead of the latest release.
A new release should be out shortly and include this bug fix.

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Please let us know if there is any further question. Thank you.

closed 2019-11-15 by RealSenseCustomerSupport