realsenseai/realsense-ros #872: Published camera_info messages not compatible with image_geometry::StereoCameraModel

tykurtz, 2019-07-31:
The frame_id of the camera_info message published on `/camera/infra2/camera_info` is camera_infra2_optical_frame. This is not compatible with the `image_geometry::StereoCameraModel` class as [StereoCameraModel::fromCameraInfo()](http://docs.ros.org/melodic/api/image_geometry/html/c++/stereo__camera__model_8cpp_source.html#l00027) makes the assumption that the left and right camera_info messages are in the same frame. So modifying the second camera to use the first camera frame would be more in line with ROS convention.

Without digging into the realsense-ros code, my hacky fix was to subscribe, modify the frame and projection matrix, and republish the modified camera_info as below, but an upstream fix would be preferred if posible.

```python
    def on_camera_info_received(self, camera_info):
        modified_camera_info = camera_info
        modified_camera_info.header.frame_id = 'camera_infra1_optical_frame'
        projection_matrix = list(modified_camera_info.P)
        projection_matrix[3] = projection_matrix[0] * .0499728  # fx * baseline
        modified_camera_info.P = projection_matrix

        self.right_camera_info_publisher.publish(modified_camera_info)
```

doronhi, 2019-08-06:
I was not aware of that convention. It means that there shouldn't be published a static transform for infra2 nor a topic named `/camera/extrinsics/depth_to_infra2`
Basically it means that infra2 should not be treated as a separate sensor.
It means an interface change.
I realize the logic in this but could you refer me to the official convention doc?

stwirth, 2019-08-06:
That spec can be found in the comment of the CameraInfo message definition:
http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/CameraInfo.html

tykurtz, 2019-08-06:
Additionally to @stwirth 's comment, for me the convention is also being compatible with [stereo_imag_proc](http://wiki.ros.org/stereo_image_proc). The expectation I have is that with the appropriate topic remapping, one could run [stereo_image_proc.launch](https://github.com/ros-perception/image_pipeline/blob/melodic/stereo_image_proc/launch/stereo_image_proc.launch) by streaming monochrome images + camera info without any hiccups. Currently, you are going to get asserts failed [in the point cloud nodelet](https://github.com/ros-perception/image_pipeline/blob/melodic/stereo_image_proc/src/nodelets/point_cloud2.cpp#L165) and [in the disparity nodelet](https://github.com/ros-perception/image_pipeline/blob/melodic/stereo_image_proc/src/nodelets/disparity.cpp#L176) due to the frame mismatch. I run into this issue in particular as I'm benchmarking different methods of disparity estimation to improve upon the on-board 3D reconstruction and wanted to reuse the point_cloud2 nodelet.

>It means that there shouldn't be published a static transform for infra2 nor a topic named /camera/extrinsics/depth_to_infra2
Basically it means that infra2 should not be treated as a separate sensor.
It means an interface change.

It would still be a separate and proper sensor, just with the optical center now located in infra1's frame. But I do see your point about no longer being able to treat infra2 as an isolated monocular camera. If someone incorrectly assumed you could just use the frame_id and the intrinsic matrix from infra2's camera_info, it would be incorrect.The consistency here is that this is true for all ROS stereo cameras.

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

Hi @tykurtz,

Did the fix https://github.com/IntelRealSense/realsense-ros/pull/901 solve the issue? Please let us know if you need any further assistance with this matter.

Thank you!

closed 2019-12-24 by RealSenseCustomerSupport