realsenseai/realsense-ros #873: Issue with Realsense D435 and T265  On Launch :An error has occurred during frame callback: Error occured during execution of the processing block

SSusantAchary, 2019-08-01:
Note:
Environment============
Camera : D435 and T265
Os Version - Ubuntu 18.04 LTS
Processor -Jetson Nano
RAM- 4GB
RealSense ROS v2.2.7
Running with LibRealSense v2.21.0
Built with python 2.7
Connected to - 5V supply
========================
Realsense-ros  built used : https://github.com/IntelRealSense/realsense-ros

even after specifying the device serial number and devices being connected to USB3.0 , error seems to occur, if i plug either one of them both seems to work individually, but NOT simultaneously.
Commands tried:(failed)
1. roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=<serial number of the first camera> serial_no_camera2:=<serial number of the second camera>

2. roslaunch realsense2_camera rs_d400_and_t265.launch

[image]
[image]

Expected Output:
On launching the above command should show the output the view as it ideally performs in realsense-viewer
[image]

SSusantAchary, 2019-08-02:
Solved

doronhi, 2019-08-06:
If you have the time, could you shortly share how was it solved?

SSusantAchary, 2019-08-06:
I was having problem of getting both the camera aligned on the pre- generated map to localize on Rviz, i assumed the above error message is causing an error.

But i solved the problem by adding a "static transform publisher" for both the camera's,

"Issue still persist" . i only closed it, as it  was not causing any issue in my localisation and navigation, as i identified later.

May be the issue is because of over flow of data from D435 and T265 to Jetson Nano.

Thanks doronhi.

doronhi, 2019-08-11:
I'm not sure I understand either the issue or the solution...
`rs-d40_and_t265.launch` includes a `static_transform_publisher` so I don't see how adding one changed anything. I hope that when running rs-d40_and_t265.launch you specified the serial numbers although the "error during frame callback..." message that you got could indeed be related to machine's processing limitations..
Anyway, I'm glad it worked out for you at end,
Thank you for sharing.

SSusantAchary, 2019-08-11:
Sorry for missing clarity, let me break down and explain.
Issue:
1. I had generated a occupancy map withD435 and T265  Camera's using RtabMap library  (it generates both 3d point cloud and 2d occupancy map) of an environment.

2. i was trying to align both camera's transformation with generated occupancy map(i.e localization/position both camera inside map as if both camera are attached to a robot)., but "rs_d400_and_t265.launch" has a static transformer which aligns t265_to_d400 (means this transformation is having in between both camera's), a link was missing between camera transformation and occupancy map .

[image]

3. i was trying this because i wanted to track the position of both camera movement in the map, which help in localizing(as T265 has localization capability) the robot(travel path of robot ) in the map.

{"4"}. I assumed the above error (An error has occurred during frame callback: Error occured during execution of the processing block) was preventing the transformation from camera's to map.
BUT it was not.

How i solved:
1. Wrote a simple transformation from both the  camera's  to map. (got idea from this https://github.com/IntelRealSense/realsense-ros/issues/749).

And You are correct the Issue is because of Hardware Processing capability, issue still continues.(i am ignoring this error as my camera's are localized and movement of robot happens smoothly ). 
 
Hope this helped.

closed 2019-08-11 by SSusantAchary