realsenseai/realsense-ros #874: does occupancy-mapping branch work well?

jungladicitta, 2019-08-02:
I have deleted my previous realsense-ros package, and did git clone from occupancy-mapping branch. And I was trying to catkin_make, and i got these errors:
`/home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:10:38: error: class ‘realsense2_camera::T265RealsenseNode’ does not have any field named ‘_wo_snr’
                                      _wo_snr(dev.first<rs2::wheel_odometer>()),
                                      ^
/home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:10:56: error: ‘wheel_odometer’ is not a member of ‘rs2’
                                      _wo_snr(dev.first<rs2::wheel_odometer>()),
                                                        ^
/home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:10:56: error: ‘wheel_odometer’ is not a member of ‘rs2’
/home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:10:77: error: no matching function for call to ‘rs2::device::first()’
                                      _wo_snr(dev.first<rs2::wheel_odometer>()),
                                                                             ^
In file included from /home/r2/catkin_ws/install_isolated/include/librealsense2/hpp/rs_record_playback.hpp:8:0,
                 from /home/r2/catkin_ws/install_isolated/include/librealsense2/hpp/rs_context.hpp:8,
                 from /home/r2/catkin_ws/install_isolated/include/librealsense2/rs.hpp:9,
                 from /home/r2/catkin_ws/src/realsense-ros/realsense2_camera/include/../include/realsense_node_factory.h:11,
                 from /home/r2/catkin_ws/src/realsense-ros/realsense2_camera/include/base_realsense_node.h:6,
                 from /home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/t265_realsense_node.h:3,
                 from /home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:1:
/home/r2/catkin_ws/install_isolated/include/librealsense2/hpp/rs_device.hpp:52:11: note: candidate: template<class T> T rs2::device::first() const
         T first() const
           ^
/home/r2/catkin_ws/install_isolated/include/librealsense2/hpp/rs_device.hpp:52:11: note:   template argument deduction/substitution failed:
/home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:10:77: error: template argument 1 is invalid
                                      _wo_snr(dev.first<rs2::wheel_odometer>()),
                                                                             ^
/home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp: In member function ‘void realsense2_camera::T265RealsenseNode::initializeOdometryInput()’:
/home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:35:10: error: ‘_wo_snr’ was not declared in this scope
     if (!_wo_snr.load_wheel_odometery_config(wo_calib))
          ^
/home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp: In member function ‘void realsense2_camera::T265RealsenseNode::odom_in_callback(const ConstPtr&)’:
/home/r2/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:68:5: error: ‘_wo_snr’ was not declared in this scope
     _wo_snr.send_wheel_odometry(0, 0, velocity);`

how can i fix it?

and also readme.md seems incorrect

doronhi, 2019-08-06:
@schmidtp1 , any idea?

schmidtp1, 2019-08-06:
Hi @jungladicitta, what LRS version are you using? The occupancy mapping branch is not actively maintained and not up to date (but was working in the past). I'd have to check for the latest LRS version and any possible changes.
What is it that seems incorrect to you in the README.md?

jungladicitta, 2019-08-07:
Hello @schmidtp1,
I am using latest version of librealsense 2.25.0 from master branch.
Recently, I fully have re-installed entire ros and librealsense, updated gcc to 9 version and cmake to 3.15.1. Now, i am not getting above errors when i am doing catkin_make.

But, I still have another problem. When i am running 
`roslaunch occupancy occupancy_live_rviz.launch`
i get randomly 1 of these 4 cases:
1) everything is not working
[image]
2) only t265 camera is working
[image]
3) t265 is working, d435 is freezing, PointCloud shows at the start, images is slowly updating
[image]
4) everything works fine, at the beginning, map is drawn well, but then seems like d435 become disconnected
[image]

about README.md, i mean: 1) why there is written "catkin build", not catkin_make; 2) "git clone <URL_TO_REPOSITORY>" without indicating occupancy-mapping branch

closed 2019-09-06 by jungladicitta

tommasomac, 2019-10-24:
Hello. Did you or anyone find a solution for this? I get the same problems, probably because the two cameras fail when they are launched together, but I could not find a solution.

gezabohus, 2020-09-21:
> Hello. Did you or anyone find a solution for this? I get the same problems, probably because the two cameras fail when they are launched together, but I could not find a solution.

You might want to check out http://wiki.ros.org/timed_roslaunch. I have used it to put a few between launching the two cameras.