realsenseai/realsense-ros #875: Setting log4j.logger.ros=DEBUG results in recursive print statement

Kaju-Bubanja, 2019-08-05:
As the title says, setting log4j.logger.ros=DEBUG results in recursive print statement and later on shutdown of the node. Here is the beginning of the log.
```
... logging to /home/satco/.ros/log/91b0202c-b75b-11e9-a2d6-e094677b42a6/roslaunch-satco-Lenovo-ideapad-Y700-15ISK-12833.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://satco-Lenovo-ideapad-Y700-15ISK:33289/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: decimation,spatia...
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/topic_odom_in: camera/odom_in
 * /camera/realsense2_camera/unite_imu_method: none
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [12843]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 91b0202c-b75b-11e9-a2d6-e094677b42a6
process[rosout-1]: started with pid [12854]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [12861]
process[camera/realsense2_camera-3]: started with pid [12862]
[DEBUG] [1564993953.297394945]: remap: __name => realsense2_camera_manager
[DEBUG] [1564993953.297463485]: remap: __log => /home/satco/.ros/log/91b0202c-b75b-11e9-a2d6-e094677b42a6/camera-realsense2_camera_manager-2.log
process[camera/color_rectify_color-4]: started with pid [12863]
[DEBUG] [1564993953.303777666]: Creating ClassLoader, base = nodelet::Nodelet, address = 0x5564d0cd2890
process[camera/points_xyzrgb_hw_registered-5]: started with pid [12864]
Warning: recursive print statement has occurred.  Throwing out recursive print.
Warning: recursive print statement has occurred.  Throwing out recursive print.
Warning: recursive print statement has occurred.  Throwing out recursive print.
Warning: recursive print statement has occurred.  Throwing out recursive print.
Warning: recursive print statement has occurred.  Throwing out recursive print.
Warning: recursive print statement has occurred.  Throwing out recursive print.
Warning: recursive print statement has occurred.  Throwing out recursive print.
Warning: recursive print statement has occurred.  Throwing out recursive print.
Warning: recursive print statement has occurred.  Throwing out recursive print.
Warning: recursive print statement has occurred.  Throwing out recursive print.

```

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@Kaju-Bubanja Sorry for late response. Did you still have issue with the latest SDK and ROS wrapper? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@Kaju-Bubanja Could you please update?Thanks!

Kaju-Bubanja, 2020-01-02:
I have no longer access to a camera and can not test this anymore.

closed 2020-01-02 by Kaju-Bubanja