realsenseai/realsense-ros #879: Dumping the dynamic parameters and then loading them again throws exception

Kaju-Bubanja, 2019-08-06:
Saving the dynamic parameters into a file for example of the temporal filter: `rosrun dynamic_reconfigure dynparam dump /camera/temporal test.yaml` and then loading it again `rosrun dynamic_reconfigure dynparam load /camera/temporal test.yaml` throws following error:
`[ERROR] [1565078492.496547243]: Exception thrown while processing service call: Given value -1is outside valid range!
` The file looks like this:

```
!!python/object/new:dynamic_reconfigure.encoding.Config
dictitems:
  filter_smooth_alpha: 0.4000000059604645
  filter_smooth_delta: 20
  frames_queue_size: 16
  groups: !!python/object/new:dynamic_reconfigure.encoding.Config
    dictitems:
      filter_smooth_alpha: 0.4000000059604645
      filter_smooth_delta: 20
      frames_queue_size: 16
      groups: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      holes_fill: 3
      id: 0
      name: Default
      parameters: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      parent: 0
      state: true
      stream_filter: 1
      stream_format_filter: 1
      stream_index_filter: -1
      type: ''
    state: []
  holes_fill: 3
  stream_filter: 1
  stream_format_filter: 1
  stream_index_filter: -1
state: []

```

Is there a way to set the needed dynamic parameters with rqt_reconfigure and then save them and reload them on startup?

doronhi, 2019-08-13:
It seems to be a bug in librealsense, where the minimum value for "stream_index_filter" is set to 0 and it's default value to -1 (meaning any index)
I'll submit a PR to librealsense. Until such time, you can manually set stream_index_filter to 0 in the test.yaml file (assuming you are filtering depth or RGB. If you filter infra1 or infra2, set it to either 1 or 2 accordingly).

doronhi, 2019-08-13:
Add [PR#4631](https://github.com/IntelRealSense/librealsense/pull/4631) to librealsense.

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

The issue has been addressed in RealSense SDK build 2.29.
Closing this ticket unless there is any problem.
Thank you.

closed 2019-11-01 by RealSenseCustomerSupport