realsenseai/realsense-ros #880: t265 imu unite issues

vivekalgorythma, 2019-08-06:
I have been trying to use the IMU unite methods for VIO and they seem to not work. When i use the copy method, there is a timestamp issue, and linear interpolation only gives a 63Hz output. 

In case of the copy, i have noticed that everytime there is a acceleration update to the message, the timestamp is older than the previous message and the sequence remains the same. I have verified it using the python code as shown below 

`#!/usr/bin/env python

import rospy

from std_msgs.msg import Header
from sensor_msgs.msg import Imu

class timestamp_continuity_checker:
    def __init__(self):
        rospy.init_node("ts_continuity_checker")
        rospy.Subscriber("/cam_odom/imu", Imu, self.imu_callback)
        self.imuPub = rospy.Publisher("/imu", Imu, queue_size=100)
        self.cnt = 0
        self.prevTS = Header()
        self.prevMsg = Imu()
        rospy.spin()

    def imu_callback(self, msg):
        if self.checker(msg.header) == 0:
            self.imuPub.publish(msg)
        else:
            print "Curr GyroX: ", msg.angular_velocity.x, " prev GyroX: ", self.prevMsg.angular_velocity.x
            print "Curr GyroY: ", msg.angular_velocity.y, " prev GyroY: ", self.prevMsg.angular_velocity.y
            print "Curr GyroZ: ", msg.angular_velocity.z, " prev GyroZ: ", self.prevMsg.angular_velocity.z

            print "Curr AccelX: ", msg.linear_acceleration.x, " prev AccelX: ", self.prevMsg.linear_acceleration.x
            print "Curr AccelY: ", msg.linear_acceleration.y, " prev AccelY: ", self.prevMsg.linear_acceleration.y
            print "Curr AccelZ: ", msg.linear_acceleration.z, " prev AccelZ: ", self.prevMsg.linear_acceleration.z

        self.prevMsg = msg
        return

    def checker(self, header):
        prevSec = self.prevTS.stamp.to_sec()
        currSec = header.stamp.to_sec()
        self.prevTS = header
        diff = currSec - prevSec
        if (diff <= 0):
            self.cnt += 1
            print "Got bad time at: ", currSec, " prev ts: ", prevSec, " at count: ", self.cnt, " with diff: ", diff * -1000, " seq_curr: ", header.seq, " seq_prev: ", self.prevTS.seq
            return -1
        return 0

tcc = timestamp_continuity_checker()`

The output of the code is shown below 

`Got bad time at:  1565014604.71  prev ts:  1565014604.71  at count:  1  with diff:  8.8951587677  seq_curr:  825  seq_prev:  825
Curr GyroX:  0.00106529693585  prev GyroX:  0.00106529693585
Curr GyroY:  -0.00532648479566  prev GyroY:  -0.00532648479566
Curr GyroZ:  0.0  prev GyroZ:  0.0
Curr AccelX:  0.0566278845072  prev AccelX:  0.0566278845072
Curr AccelY:  9.77774906158  prev AccelY:  9.77774906158
Curr AccelZ:  -0.755038499832  prev AccelZ:  -0.755038499832
Got bad time at:  1565014604.72  prev ts:  1565014604.73  at count:  2  with diff:  8.17942619324  seq_curr:  829  seq_prev:  829
Curr GyroX:  0.00319589092396  prev GyroX:  0.00319589092396
Curr GyroY:  -0.00532648479566  prev GyroY:  -0.00532648479566
Curr GyroZ:  0.0  prev GyroZ:  0.0
Curr AccelX:  0.0566278845072  prev AccelX:  0.0566278845072
Curr AccelY:  9.77774906158  prev AccelY:  9.77774906158
Curr AccelZ:  -0.773914396763  prev AccelZ:  -0.755038499832
Got bad time at:  1565014604.74  prev ts:  1565014604.74  at count:  3  with diff:  7.46417045593  seq_curr:  833  seq_prev:  833
Curr GyroX:  0.00106529693585  prev GyroX:  0.00106529693585
Curr GyroY:  -0.00639178184792  prev GyroY:  -0.00639178184792
Curr GyroZ:  -0.0021305938717  prev GyroZ:  -0.0021305938717
Curr AccelX:  0.0377519242465  prev AccelX:  0.0566278845072
Curr AccelY:  9.77774906158  prev AccelY:  9.77774906158
Curr AccelZ:  -0.773914396763  prev AccelZ:  -0.773914396763
Got bad time at:  1565014604.75  prev ts:  1565014604.76  at count:  4  with diff:  6.74772262573  seq_curr:  837  seq_prev:  837
Curr GyroX:  0.0021305938717  prev GyroX:  0.0021305938717
Curr GyroY:  -0.00319589092396  prev GyroY:  -0.00319589092396
Curr GyroZ:  -0.0021305938717  prev GyroZ:  -0.0021305938717
Curr AccelX:  0.0377519242465  prev AccelX:  0.0377519242465
Curr AccelY:  9.79662322998  prev AccelY:  9.77774906158
Curr AccelZ:  -0.755038499832  prev AccelZ:  -0.773914396763
Got bad time at:  1565014604.77  prev ts:  1565014604.77  at count:  5  with diff:  6.03294372559  seq_curr:  841  seq_prev:  841
Curr GyroX:  0.00532648479566  prev GyroX:  0.00532648479566
Curr GyroY:  -0.00639178184792  prev GyroY:  -0.00639178184792
Curr GyroZ:  0.0  prev GyroZ:  0.0
Curr AccelX:  0.0377519242465  prev AccelX:  0.0377519242465
Curr AccelY:  9.79662322998  prev AccelY:  9.79662322998
Curr AccelZ:  -0.773914396763  prev AccelZ:  -0.755038499832
Got bad time at:  1565014604.78  prev ts:  1565014604.79  at count:  6  with diff:  5.31721115112  seq_curr:  845  seq_prev:  845
Curr GyroX:  0.0021305938717  prev GyroX:  0.0021305938717
Curr GyroY:  -0.00532648479566  prev GyroY:  -0.00532648479566
Curr GyroZ:  -0.0042611877434  prev GyroZ:  -0.0042611877434
Curr AccelX:  0.0377519242465  prev AccelX:  0.0377519242465
Curr AccelY:  9.77774906158  prev AccelY:  9.79662322998
Curr AccelZ:  -0.773914396763  prev AccelZ:  -0.773914396763
Got bad time at:  1565014604.8  prev ts:  1565014604.8  at count:  7  with diff:  4.60052490234  seq_curr:  849  seq_prev:  849
Curr GyroX:  0.00532648479566  prev GyroX:  0.00532648479566
Curr GyroY:  -0.00319589092396  prev GyroY:  -0.00319589092396
Curr GyroZ:  0.00106529693585  prev GyroZ:  0.00106529693585
Curr AccelX:  0.0377519242465  prev AccelX:  0.0377519242465
Curr AccelY:  9.79662322998  prev AccelY:  9.77774906158
Curr AccelZ:  -0.773914396763  prev AccelZ:  -0.773914396763
Got bad time at:  1565014604.82  prev ts:  1565014604.82  at count:  8  with diff:  8.88299942017  seq_curr:  854  seq_prev:  854
Curr GyroX:  0.00106529693585  prev GyroX:  0.00106529693585
Curr GyroY:  -0.00745707890019  prev GyroY:  -0.00745707890019
Curr GyroZ:  0.00319589092396  prev GyroZ:  0.00319589092396
Curr AccelX:  0.0377519242465  prev AccelX:  0.0377519242465
Curr AccelY:  9.79662322998  prev AccelY:  9.79662322998
Curr AccelZ:  -0.773914396763  prev AccelZ:  -0.773914396763
`
As seen above this happens only when the accel is updated, the sequence no is the same and the timestamp is older. 

When i use linear interpolation, i get IMU at 63Hz, which is too low for VIO. But the timestamp issue is gone.

Is there a way to fix this ?

doronhi, 2019-08-08:
What type of device are you using? D435i? T265?

vivekalgorythma, 2019-08-13:
@doronhi Im using the T265, on a USB 2.0 bus

vivekalgorythma, 2019-08-15:
@doronhi any updates on this issue ?

TarekTaha, 2019-08-15:
Same issue here, any solution?

freaklyNice, 2019-09-04:
Same issue. The reason seemly to be GYRO and ACCEL streams don't arrive in order.

doronhi, 2019-09-10:
Sorry for the delay.
The problem is this: There is a different processing time for the gyro and accel back in the device. As a result, the gyro is sent after a delay T and the accel is sent after a delay of T+8 (approximately).
Therefor, for example, at system time t we receive a gyro message taken at hardware time T and at system time t+1 we receive an accel message taken at T-7.
They do not arrive according to their original order.
A possible solution is to send the data only when gyro is updated. That will result in gyro fps.

rgreid, 2019-10-04:
Any update on this issue?  We tried working around it by publishing the acceleration part of the pose stream. However noted that the acceleration lags by ~18 ms. Does the T+8 delay also affect the T265's onboard filter, or is it compensated for?

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

This thread has been opened for quite a while.  There has been updates to the unite_imu_method here:  https://github.com/IntelRealSense/realsense-ros/pull/1010

Also including the comments provided by doronhi regarding possible workaround.  Let us know if there is anything else needed on this thread.

closed 2020-02-04 by RealSenseCustomerSupport