realsenseai/realsense-ros #881: t265 and d435 mount, extrinsic parameter doubt

svabhishek29, 2019-08-06:
Hi, 

i was looking at the extrinsic parameter for the mount for the t265 and d435 cameras, the 

> <origin xyz="0.009 0.021 0.027" rpy="0.000 -0.018 0.005"/> 

line says the depth camera is at xyz offset and rpy degrees from the tracker camera, is this with respect to center of the depth camera?

if so how do i get it wrt to the rgb camera on d435?

thanks

doronhi, 2019-08-08:
can you use the topic:
`/camera/extrinsics/depth_to_color`?
the transformation is also available in the /tf_static.

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Hi @svabhishek29

Were you able to utilize the topic that doronhi provided to get the information you needed?

Thanks

closed 2019-10-11 by RealSenseCustomerSupport