realsenseai/realsense-ros #882: Kinetic中安装realsense-ros包catkin_make:‘filter’ is not a member of ‘rs2’

wangqingdong1, 2019-08-08:
Kinetic中安装realsense-ros包catkin_make:‘filter’ is not a member of ‘rs2’，以及template argument 1 is invalid，‘wheel_odometer’ in namespace ‘rs2’ does not name a type，class ‘realsense2_camera::T265RealsenseNode’ does not have any field named ‘_wo_snr’等问题。

ahuizxc, 2019-09-02:
please try to update your librealsense to the newest version:)

RealSenseCustomerSupport, 2019-11-04:
----------------------------------------------

@wangqingdong1 Did you get chance to try the latest version to see if you still have issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@wangqingdong1 Sorry that we didn't get response from you for weeks. Will close it at this point. If you still have questions/issues, please feel free to create another one. Thanks!

closed 2019-12-02 by RealSenseCustomerSupport