realsenseai/realsense-ros #885: Broken xacro support for multiple cameras

tykurtz, 2019-08-08:
There are a couple issues that I noticed with the URDF when trying to incorporate multiple camera models onto a robot.

Here are the tf trees after when running both the view_d435_model.launch and the ROS drivers.
```
# Example realsense drivers tf tree for D435
arg tf_prefix := ""
${tf_prefix}_link
|_ ${tf_prefix}_depth_frame
    |_ ${tf_prefix}_depth_optical_frame
|_ ${tf_prefix}_color_frame
    |_ ${tf_prefix}_color_optical_frame
|_ ${tf_prefix}_infra1_frame
    |_ ${tf_prefix}_infra1_optical_frame
|_ ${tf_prefix}_infra2_frame
    |_ ${tf_prefix}_infra2_optical_frame
```

```
# URDF tf tree
camera_link
|_ camera_depth_frame
    |_ camera_color_frame
        |_ camera_color_optical_frame
    |_ camera_depth_optical_frame
    |_ camera_left_ir_frame
        |_ camera_left_ir_optical_frame
    |_ camera_right_ir_frame
        |_ camera_right_ir_optical_frame
```
- [ ] Redundant frames are being published by the node if the urdf is included. This issue is already documented and can be addressed through #777 
- [ ] Inconsistent frame naming notation between the URDF and realsense driver. On the drivers, the monochrome cameras are called infra1 and infra2, and in the URDF it's left and right.
- [ ] The sensor frames derive from the depth frame for the URDF, and from the camera_link frame on the drivers.
- [ ] There is no ${prefix} argument as mentioned [here](https://github.com/IntelRealSense/realsense-ros/issues/494#issuecomment-439451045). This limits the usefulness of having a xacro at all. 

These changes should be straightforward, so I don't mind contributing. I just want to make clear the preferred changes before starting. To me, the left/right notation makes more sense, and is more in line with the `/stereo/left/` and `/stereo/right/` ROS namespace convention. However, making this change would break existing interfaces. I also prefer deriving from camera_link instead of the depth frame, so I would modify the URDF to match that notation.

saikishor, 2019-10-21:
FYI
https://github.com/IntelRealSense/realsense-ros/pull/957

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

Thanks @saikishor's contribution.

closed 2019-12-05 by RealSenseCustomerSupport