realsenseai/realsense-ros #887: Hardware Notification:RT IC2 Config error,1.56535e+12,Error,Hardware Error

dario-ljubic, 2019-08-09:
I am experiencing a strange behaviour using my cameras. I am aware that this issue is reported before, but I did not see it was resolved in some cases. My usb port is 3.0 and I checked the output of an `lsusb` command to see if the cameras are recognized as 3.0. They are.

```
09/08 10:58:23,358 WARNING [140708967085824] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1565341103.358435314]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 09/08 10:58:23,400 WARNING [140709254797056] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1565345293.220079163]: Hardware Notification:RT IC2 Config error,1.56535e+12,Error,Hardware Error
[ERROR] [1565345293.220119608]: Hardware Reset is needed. use option: initial_reset:=true
 09/08 12:08:14,535 ERROR [140481189578496] (types.h:243) xioctl(VIDIOC_DQBUF) failed for fd: 30 Last Error: No such device
 09/08 12:08:14,535 ERROR [140481199601408] (types.h:243) xioctl(VIDIOC_DQBUF) failed for fd: 22 Last Error: No such device
 09/08 12:08:14,536 ERROR [140481189578496] (backend-v4l2.cpp:1236) xioctl(VIDIOC_DQBUF) failed for fd: 30 Last Error: No such device
[ WARN] [1565345294.536993073]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for fd: 30 Last Error: No such device,1.56535e+12,Error,Unknown Error
 09/08 12:08:14,537 ERROR [140481199601408] (backend-v4l2.cpp:1236) xioctl(VIDIOC_DQBUF) failed for fd: 22 Last Error: No such device
[ WARN] [1565345294.537231118]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for fd: 22 Last Error: No such device,1.56535e+12,Error,Unknown Error
 09/08 12:08:14,541 ERROR [140481365071616] (types.h:243) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/08 12:08:14,541 ERROR [140481365071616] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/08 12:08:14,627 ERROR [140481490863872] (types.h:243) Cannot open '/dev/video3 Last Error: No such device
 09/08 12:08:14,627 ERROR [140481490863872] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: Cannot open '/dev/video3 Last Error: No such device
 09/08 12:08:15,541 ERROR [140481365071616] (types.h:243) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/08 12:08:15,541 ERROR [140481365071616] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/08 12:08:15,627 ERROR [140481490863872] (types.h:243) Cannot open '/dev/video3 Last Error: No such device
 09/08 12:08:15,627 ERROR [140481490863872] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: Cannot open '/dev/video3 Last Error: No such device
 09/08 12:08:16,541 ERROR [140481365071616] (types.h:243) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/08 12:08:16,541 ERROR [140481365071616] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 09/08 12:08:16,628 ERROR [140481490863872] (types.h:243) Cannot open '/dev/video3 Last Error: No such device
 09/08 12:08:16,628 ERROR [140481490863872] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: Cannot open '/dev/video3 Last Error: No such device
 09/08 12:08:17,312 ERROR [140481524434688] (types.h:243) Cannot identify '/dev/video7 Last Error: No such file or directory
 09/08 12:08:17,312 ERROR [140481524434688] (types.h:243) Cannot identify '/dev/video8 Last Error: No such file or directory
 09/08 12:08:17,312 ERROR [140481524434688] (types.h:243) Cannot identify '/dev/video6 Last Error: No such file or directory
[ERROR] [1565345297.312719139]: The device has been disconnected!
 09/08 12:08:17,347 ERROR [140481524434688] (types.h:243) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 09/08 12:08:17,865 ERROR [140709513979648] (types.h:243) Cannot identify '/dev/video7 Last Error: No such file or directory
 09/08 12:08:17,865 ERROR [140709513979648] (types.h:243) Cannot identify '/dev/video8 Last Error: No such file or directory
 09/08 12:08:17,865 ERROR [140709513979648] (types.h:243) Cannot identify '/dev/video6 Last Error: No such file or directory
 09/08 12:08:22,347 ERROR [140481524434688] (types.h:243) Cannot identify '/dev/video7 Last Error: No such file or directory
 09/08 12:08:22,347 ERROR [140481524434688] (types.h:243) Cannot identify '/dev/video8 Last Error: No such file or directory
```

| Camera Model                       | D435 | 
| Firmware Version                   |Tried both 5.10.13 and 5.11.6.250 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-55-generic    | 
| Platform                                 |  Intel® NUC Kit NUC7i7DNHE  |
| librealsense commit                            | f112ed78  | 
|realsense ros wrapper    |    31fa75ec   |

I am using 2 cameras and have color fps set at 15 Hz, and infra and depth fps at 30 Hz. Resolution is at 848x480 and I have 2 D435 cameras.

ilidar, 2019-08-13:
Having the same issue using just one SR300 camera on Thinkpad X1 Extreme.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

please ensure you install the required Realsense DKMS kernel driver package on your ubuntu PC
you can refer to below link for detail
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

in parallel, to ensure your camera is still in good shape, you can connect them to any windows 10 PC and run the intel.realsense.viewer.exe to stream them
https://github.com/IntelRealSense/librealsense/releases/tag/v2.29.0

for Win10 OS, no additional driver package is required.

closed 2020-02-24 by RealSenseCustomerSupport