realsenseai/realsense-ros #889: Error handling and device reset

paulbovbel, 2019-08-12:
The driver has a hardcoded list of error strings that will trigger a device reset:

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp#L164

Empirically, I've seen at least two more errors that cause the driver to stop generating frames, and are recoverable via reset:

https://github.com/IntelRealSense/librealsense/blob/d19829788008b8e000870895a068f0c43d58895a/src/ds5/ds5-private.h#L678
```
            { usb_scp_overflow,             "USB SCP overflow" },
            { rec_error,                    "REC error" },
```
Is there any reason not to attempt a device reset on any hardware error?

MartyG-RealSense, 2019-08-12:
For a lot of users, who desire to record as many good frames as possible, they want to reset the stream as quickly as possible if a USB error occurs (which is why hardware_reset is useful for automating the reset).

Some software may take the approach of taking steps to try to correct the problem before resorting to a reset or shutdown.  For example, if a 400 Series camera has a very high operating temperature then its Laser Power may be halved.  If the very high temperature continues for a certain period then the camera is shut off altogether.

Given how errors such as overflow continually cascade once they occur, displaying over and over, reset rather than attempting autocorrect actions is likely the best course of action.  There is an old saying that goes "Least said, soonest mended".     :)

paulbovbel, 2019-08-19:
@MartyG-RealSense, sorry, i'm unclear, what's the suggested course of action?

MartyG-RealSense, 2019-08-19:
You asked: "Is there any reason **not** to attempt a device reset on any hardware error?"  My opinion was that it is best to reset as soon as possible after an error occurs instead of just let it run on and generate a mounting cascade of errors.

paulbovbel, 2019-08-19:
Agreed. As you can see from my link, there's a list of errors hardcoded into the ROS driver that will trigger a rest. My question is - should this list be expanded, or would it make sense just to reset on any error by default?

I'm happy to prepare a patch, though I'd lean towards the latter, and I'd love some guidance from the maintainers of this repository since I'd hope to get it merged.

MartyG-RealSense, 2019-08-19:
The topic is outside of my knowledge.  I'll link into this discussion one of the main Intel ROS guys @doronhi since it's a ROS related question.

doronhi, 2019-09-02:
Both error messages, about overflow and REC_ERROR, require some system wide modification. Reducing frame rate, resolution, increasing computing power, etc. Without one, a simple reset will only solve the problem temporarily, until a new pile of data will be accumulated and a new overflow message will appear.
Hardware reset, in my opinion, is meant to overcome firmware and hardware problems that are yet to be fixed and could not be handled otherwise.
If other errors of that sort still exist I think they should be entered into the fixed list inside the wrapper.

paulbovbel, 2019-09-02:
> Without one, a simple reset will only solve the problem temporarily, until a new pile of data will be accumulated and a new overflow message will appear.

That makes sense. I assume that if this error is hit during some edge-case load on the host system, it would still be useful to perform a 'simple reset' and get the sensor operational again. It would be up to a monitoring system to keep track of the cause/frequency of errors in a system.

Would you accept a PR to make the list of these 'reset-triggering' errors parametrized?

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@paulbovbel Is this still issue for you? Looking forward to your update. Thanks!

paulbovbel, 2019-12-13:
This is definitely still an issue. There appear to be a variety of hardware errors that are recoverable via software reset (https://github.com/IntelRealSense/realsense-ros/issues/889#issue-479723680). I have a proposed fix at https://github.com/IntelRealSense/realsense-ros/pull/946 to allow users to expand the list of errors that trigger this behavior.

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

@paulbovbel Thanks for your contribution! I saw our engineer is now looking into the PR you submitted.

closed 2020-02-03 by RealSenseCustomerSupport

paulbovbel, 2020-02-07:
The PR is still open, so this is still an issue...