realsenseai/realsense-ros #892: Flat surface with RealSense2

YannRodriguez, 2019-08-15:
Hi !
I scan a little box with realse2 camera and i've got this result :

[image]

[image]

I need to be precise, so is it normal that for a flat surface there is this kind of ondulations ?
Have you some idea how to fix it ?

stevensu1838, 2019-08-15:
Hi all,

I am a co-worker of YannRodriguez. We are scanning a box with a Realsense 435i  and reading the point cloud out of Realsense 435i on ROS side. Then we use these point cloud data to recreate the shape of the object(a box) in Unity(as you have seen in the pic in the YannRodriguez's post above).

The problem that we are facing is: 
the surface of the box we got in Unity(reconstructed from the point cloud of real sense) is not as flat as the surface of the real box that has been scanned.  
And when we only check the ROS side, we feel the surface of the box we got in Rvize(reconstructed from the point cloud of real sense) is not as flat as the surface of the real box that has been scanned, 
either. For this reason, we think the problem might happen on real sense ROS side. 

Can any of you tell us how we can solve this problem? How can we precise get the real shape of the object that is being scanned? 
Looking forward to hearing from you soon.

MartyG-RealSense, 2019-08-15:
Similar results to yours have occurred in the past, yes.  Here is an example:

[image]

And here is an image of a point cloud of a box taken from an Intel promotional web-page for the 400 Series cameras.

[image]

Dorodnic the RealSense SDK Manager offers some advice about dealing with wavy surfaces in the link below.

https://github.com/IntelRealSense/librealsense/issues/1053#issuecomment-359415769

There is also an Intel guide for tuning the camera:

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

And a guide to post-processing filters:

https://www.intel.co.uk/content/www/uk/en/support/articles/000028866/emerging-technologies/intel-realsense-technology.html

stevensu1838, 2019-08-15:
Hi @MartyG-RealSense 
I've read through Dorodnic the RealSense SDK Manager offers some advice
I am experiencing exactly the same problem.
But I am a user of real sense ROS. My question is:
As a ros user, how do I modify the settings of presets and post-processing mentioned in Dorodnic's [answer ](https://github.com/IntelRealSense/librealsense/issues/1053#issuecomment-359415769) and save a customized user preset. Do I need to do it with real sense viewer? You mentioned it is doable in code, Can you please show me how？

MartyG-RealSense, 2019-08-16:
The subject of setting post-processing filters in ROS is discussed here:

https://github.com/IntelRealSense/realsense-ros/issues/742

In regard to setting visual presets with ROS: presets take the form of a 'json' configuration file, and I found information by searching for 'json' instead of 'visual preset'.

https://github.com/IntelRealSense/realsense-ros/issues/411

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Please let us know if there is any further question related the to the original issue.
Thank you.

closed 2019-11-09 by RealSenseCustomerSupport