realsenseai/realsense-ros #893: difference between two ways of reading pointcloud with D435

stevensu1838, 2019-08-16:
Hi all,

There are two ways of reading point cloud from real sense D435 in ROS. Can any of you tell me the difference? 

Method1
roslaunch realsense2_camera rs_rgbd.launch

topic: /camera/depth_registered/points

Method2 
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud

topic: /camera/depth/color/points

sthoduka, 2019-11-06:
See https://github.com/IntelRealSense/realsense-ros/issues/851

My guess is that with rs_rgbd, the pointcloud is constructed outside the realsense node (using rgbd_launch, and the depth_image_proc nodelets). This is an organized pointcloud.

With rs_camera with 'pointcloud' filter, the pointcloud is constructed in the realsense node itself (https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp#L2044). This is an unorganized pointcloud.

I do not know if there are any other differences.

RealSenseCustomerSupport, 2020-01-13:
----------------------------------------------

Hi @stevensu1838,

@sthoduka has provided the right explanation regarding the main difference between two these ways.Please let us know if you need any further clarifications regarding this topic.

Thank you!

closed 2020-01-18 by RealSenseCustomerSupport