realsenseai/realsense-ros #894: ERROR [140395770017536] (ds5-options.cpp:82) Asic Temperature value is not valid!

nyxrobotics, 2019-08-19:
This error shows in around once in 1min.
Does anybody know how to resolve this?

device:realsense D415
environment:ubuntu16.04 + ROS kinetic
branch: commit cf8608

doronhi, 2019-09-03:
The ASIC temperature sensor is checked every 10 seconds. It can sometimes fail to return an answer and that's the meaning of the error. An increase in the frequency of the errors can mean that the firmware might fail soon and a reset will be required. This is not a definite rule so no automatic behavior was defined to handle this warning.

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

Hi @nyxrobotics,

Did you try the latest librealsense and firmware and see how it works?

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

Since this result is expected, close this. If you have any concern, feel free to re-open it.

closed 2019-12-05 by RealSenseCustomerSupport

zheng-xiao, 2021-06-30:
Hi, I have the same question. Did you solve it?

nyxrobotics, 2021-06-30:
No

teymurov27, 2022-02-15:
Hi, I have the same warning with librealsense v2.50.0

This is the terminal output. 
```bash

roslaunch realsense2_camera rs_camera.launch                                                                                                                                                                                                 
... logging to /home/thor1n/.ros/log/263e8902-8e34-11ec-bc1b-1dc07581780a/roslaunch-patika-DX-1100-2319.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:41595/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 88 lines cut ...]
[ INFO] [1644911522.885967738]: SELECTED BASE:Depth, 0
[ INFO] [1644911522.891559955]: RealSense Node Is Up!
[ WARN] [1644911522.954028095]: 
 15/02 10:52:02,954 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 10:52:03,005 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 10:52:03,056 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 10:52:03,208 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 10:52:03,259 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 10:52:03,310 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 10:52:03,361 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 10:52:03,412 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 10:52:03,463 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 15/02 10:53:22,795 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 10:57:39,026 ERROR [140487740241664] (ds5-options.cpp:88) Asic Temperature value is not valid!
 15/02 10:57:39,079 WARNING [140488296355584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/02 10:57:40,030 ERROR [140487740241664] (ds5-options.cpp:88) Asic Temperature value is not valid!
 15/02 10:57:41,033 ERROR [140487740241664] (ds5-options.cpp:88) Asic Temperature value is not valid!
```