realsenseai/realsense-ros #895: Ends with errors if run for a long time(3-days).

nyxrobotics, 2019-08-19:
After running realsense d415 for 3 days, the following error occurred.

>  18/08 23:19:51,110 ERROR [140395770017536] (ds5-options.cpp:82) Projector Temperature value is not valid!
> [ WARN] [1566137991.635896135]: Hardware Notification:Left IC2 Config error,1.56614e+12,Error,Hardware Error
> [ WARN] [1566137992.636143087]: Hardware Notification:Left IC2 Config error,1.56614e+12,Error,Hardware Error
> [ WARN] [1566137993.636364901]: Hardware Notification:Left IC2 Config error,1.56614e+12,Error,Hardware Error
> [ WARN] [1566137994.636587966]: Hardware Notification:Left IC2 Config error,1.56614e+12,Error,Hardware Error
>  18/08 23:19:54,920 WARNING [140395848931072] (types.cpp:48) set_xu(ctrl=1) failed! Last Error: Input/output error
>  18/08 23:19:54,920 ERROR [140395848931072] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Input/output error
> [ WARN] [1566137995.636788743]: Hardware Notification:Left IC2 Config error,1.56614e+12,Error,Hardware Error
>  18/08 23:20:00,935 WARNING [140395848931072] (types.cpp:48) set_xu(ctrl=1) failed! Last Error: Input/output error
>  18/08 23:20:00,935 ERROR [140395848931072] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Input/output error
>  18/08 23:20:01,111 ERROR [140395770017536] (ds5-options.cpp:82) Projector Temperature value is not valid!
>  18/08 23:20:06,951 WARNING [140395848931072] (types.cpp:48) set_xu(ctrl=1) failed! Last Error: Input/output error
>  18/08 23:20:06,951 ERROR [140395848931072] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Input/output error

After unplugging the device and waiting for a while, I reconnected and started up again.
My launch is here.

>  <launch>
>    <arg name="camera"              default="camera"/>
>    <arg name="tf_prefix"           default="$(arg camera)"/>
>    <arg name="external_manager"    default="false"/>
>    <arg name="manager"             default="realsense2_camera_manager"/>
>  
>    <arg name="serial_no"           default=""/>
>    <arg name="json_file_path"      default=""/>
>  
>    <arg name="fisheye_width"       default="640"/>
>    <arg name="fisheye_height"      default="480"/>
>    <arg name="enable_fisheye"      default="false"/>
>  
>    <arg name="depth_width"         default="640"/>
>    <arg name="depth_height"        default="480"/>
>    <arg name="enable_depth"        default="true"/>
>  
>    <arg name="infra_width"         default="640"/>
>    <arg name="infra_height"        default="480"/>
>    <arg name="enable_infra1"       default="false"/>
>    <arg name="enable_infra2"       default="false"/>
>  
>    <arg name="color_width"         default="640"/>
>    <arg name="color_height"        default="480"/>
>    <arg name="enable_color"        default="true"/>
>  
>    <arg name="fisheye_fps"         default="6"/>
>    <arg name="depth_fps"           default="90"/>
>    <arg name="infra_fps"           default="90"/>
>    <arg name="color_fps"           default="60"/>
>    <arg name="gyro_fps"            default="400"/>
>    <arg name="accel_fps"           default="250"/>
>    <arg name="enable_gyro"         default="false"/>
>    <arg name="enable_accel"        default="false"/>
>  
>    <arg name="enable_pointcloud"   default="false"/>
>    <arg name="enable_sync"         default="false"/>
>    <arg name="align_depth"         default="true"/>
>    <arg name="filters"             default=""/>
>  
>    
>  
>    
>    <arg name="rgb"                             default="color" />
>    <arg name="ir"                              default="infra1" />
>    <arg name="depth"                           default="depth" />
>    <arg name="depth_registered_pub"            default="depth_registered" />
>    <arg name="depth_registered"                default="depth_registered" unless="$(arg align_depth)" />
>    <arg name="depth_registered"                default="aligned_depth_to_color" if="$(arg align_depth)" />
>    <arg name="depth_registered_filtered"       default="$(arg depth_registered)" />
>    <arg name="projector"                       default="projector" />
[... 90 lines cut ...]
>          <remap from="depth_registered/image_rect" to="$(arg depth_registered)/sw_registered/image_rect_raw" />
>        </node>
>  
>        
>        <node pkg="nodelet" type="nodelet" name="points_xyzrgb_sw_registered"
>              args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
>          <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_throttle" />
>          <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
>          <remap from="depth_registered/image_rect" to="$(arg depth_registered_filtered)/sw_registered/image_rect_raw" />
>          <remap from="depth_registered/points"     to="$(arg depth_registered)/points" />
>        </node>
>      </group>
>  
>      <group if="$(eval depth_registered_processing and hw_registered_processing)">
>        
>        <node pkg="nodelet" type="nodelet" name="points_xyzrgb_hw_registered"
>              args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
>          <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_throttle" />
>          <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
>          <remap from="depth_registered/image_rect" to="$(arg depth_registered)/image_raw" />
>          <remap from="depth_registered/points"     to="$(arg depth_registered_pub)/points" />
>        </node>
>      </group>
>    </group>
>  </launch>

MartyG-RealSense, 2019-08-19:
As a rule of thumb when debugging, I look at the very first error in the list as a priority, as the rest after that may be problems cascading as a result of the first error that would not have occurred otherwise.

Apparently the invalid temperature issue is a long term bug that has been tricky for the RealSense developer team to pin down the cause of.

https://github.com/IntelRealSense/librealsense/issues/866

Although the 400 Series cameras can run indefinitely as long as they remain within temperature tolerances, problems with the computer hardware,

doronhi, 2019-09-03:
The error message "Left IC2 Config error" is indeed an indication that the ASIC is down. 
I will go on and add it to the list of errors that will create automatic reset in the wrapper.

RealSenseCustomerSupport, 2020-01-20:
----------------------------------------------

@nyxrobotics Did you get chance to try other D415 to see if get the issue? Any other questions? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@nyxrobotics Could you please update? If we don’t hear from you in 7 days, this issue will be closed. Thanks!

nyxrobotics, 2020-02-12:
Maybe hardware or driver issue. I close this issue.

closed 2020-02-12 by nyxrobotics