realsenseai/realsense-ros #896: realsense-viewer: error while loading shared libraries: librealsense2.so.2.25: cannot open shared object file: No such file or directory

harshitsankhla, 2019-08-19:
**OS : Ubuntu 16.04.6
ROS : Kinetic
Camera - D435**

For the past few days I have tried to build realsense library (source and debian packages) and the ros wrapper on 2 separate computers (Intel NUCs, the irony !) I am able to build both the packages but on launch (roslaunch realsense2_camera rs_rgbd.launch) I keep getting the error (most of the times, and most recently stuck here after a fresh install) - 

_/opt/ros/kinetic/lib/nodelet/nodelet: symbol lookup error: /usr/lib/x86_64-linux-gnu/librealsense2.so.2: undefined symbol: libusb_get_parent_

**Most Recent Install sequence -**

version="2.18.1-0~realsense0.568"
sudo apt-get -y install librealsense2-udev-rules=${version}
sudo apt-get -y install librealsense2-dkms=1.3.4-0ubuntu1
sudo apt-get -y install librealsense2=${version}
sudo apt-get -y install librealsense2-utils=${version}
sudo apt-get -y install librealsense2-dev=${version}
sudo apt-get -y install librealsense2-dbg=${version}

mkdir -p /rsws/src
cd rsws/src
catkin_init_workspace
git clone https://github.com/intel-ros/realsense.git
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
echo "source ~/rsws/devel/setup.bash" >> ~/.bashrc
source ~/rsws/devel/setup.bash

**Tried Workarounds -** 

1. Format the computer and build before installing any other software or package besides ROS
2. Separately install libusb-1.0.0-dev from source and apt
3. Build from source a previous version of librealsense
4. Saw a thread mentioning a conflict with Bluefox2 camera, but that should've been resolved    when I formatted the computer and tried again.

Would appreciate any kind of help to work this out.

doronhi, 2019-09-02:
Somehow, it seems that installing udev-rules messed things up.
Did you follow the installation notes [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)?
It clearly states to install librealsense2-dkms and librealsense2-utils and they will extract the rest.
It is also what been done in the [.travis file](https://github.com/IntelRealSense/realsense-ros/blob/development/.travis.yml) in the "before_install" section.
Following these instructions on a clean machine should work?
After that, try realsense-viewer. Is it working?
If that seems ok, start installing the ROS environment and the ROS wrapper following the instructions [here](https://github.com/IntelRealSense/realsense-ros)

harshitsankhla, 2019-09-04:
Well, on a clean machine the installation works fine (through sudo apt-get debian packages). I formatted 2 of my Intel NUCs and tested. Both the library and the ROS wrapper work fine.

But that should not be the goto option, to end up formatting every computer in order to install the library.

RealSenseCustomerSupport, 2019-11-05:
----------------------------------------------

Hi @harshitsankhla,

There is no any need to format the computer in order to install the library. It suffices to run
the 'ldd' command in order to find out where libusb is expected to be found.
Please let us know if you need any further clarifications regarding this topic.

Thank you!

closed 2019-11-12 by RealSenseCustomerSupport