realsenseai/realsense-ros #899: How to adjust the camera filters in the launch file

Joar93, 2019-08-22:
Hello!

First of alla great products! I use both the T265 and the D435 and i like them alot!

Im currently working on a launchscript for d435 that creates a pointcloud. This pointcloud is input to catographer. When running the script i can obseve it in RVIZ and i can change the filters settings in rqt_reconfigure. Now to my question, how can i change the settings on the filters in the launch file so that it starts up with these settings as soon as i launch the script? I cant find any good documentation on this issue? For example if i wanted to increase thedecimation filter to 4 how would i write it in the launch file?

Warm regards
Joar

Script
-------
<launch>
  <arg name="serial_no"           default="829212071528"/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="d435"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="true"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="640"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="false"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_ANY"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="false"/>

  <arg name="filters"                   default="pointcloud,decimation,spatial"/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>

  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra_width"              value="$(arg infra_width)"/>
      <arg name="infra_height"             value="$(arg infra_height)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>

 <node pkg="tf" type="static_transform_publisher" name="base_to_d435_link" args="0.57 0 0 0  0 0 base $(arg tf_prefix)_link 100" />
</launch>
------------

MartyG-RealSense, 2019-08-22:
I hope this link will be of help:

https://github.com/IntelRealSense/realsense-ros/issues/742#issuecomment-518628338

MartyG-RealSense, 2019-08-28:
Thanks muchly for the clarification about where the update was applied, @doronhi

doronhi, 2019-08-28:
Thanks @MartyG-RealSense for the link. Did it help you @Joar93 ?
It was a realsense2_camera issue and was fixed in version 2.2.8.

Joar93, 2019-08-28:
Hello! 

Thanks for the quick response!
@doronhi Yes that looks like a good solution to changing dynamic parameters! Unfortunatly im using the 2.2.5 build of the roswrappers since im having major compability issues with my jetson TX2 and realsense software and i have only been able to get the older realsense SDK to work. However i noticed in the link that you can simply launch it as a node like this!
```
<node name="configure_temporal_filter" pkg="dynamic_reconfigure" type="dynparam"
	args="set_from_parameters camera/temporal">
    <param name="filter_smooth_delta" value="50" />
    <param name="filter_smooth_alpha" value="0.1"/>
    <param name="holes_fill"          value="3"  />
  </node>
```
This is a good solution for now! Thanks again for the quick reply and good work on the ROS-wrapper @doronhi! Also thanks to @MartyG-RealSense for providing the link! Hopefully the SDK will work better with the jetson hardware in the future!

closed 2019-08-28 by Joar93