realsenseai/realsense-ros #900: How to adjust the camera exposure by topic

WangyuHIT, 2019-08-23:


doronhi, 2019-08-28:
Could you elaborate a little...?

wikpunGitHub, 2019-10-29:
Hi, I have the same question. Is there a way to adjust the "set auto exposure" setting in launch file or node parameter rather than using rqt_reconfigure?

doronhi, 2019-10-29:
You can add the following lines to your launch file before the section of launching realsense2_camera node:
```
<rosparam>
  /camera/rgb_camera/auto_exposure_roi/left: 30
  /camera/rgb_camera/auto_exposure_roi/right: 350
</rosparam>
```
(you can check the precise string using rqt_reconfigure)

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

Hi @ WangyuHIT,

Has the comment provided by @doronhi helped to find an answer to your question? Please let us know if you need any further assistance with this matter.

Thank you!

closed 2019-12-24 by RealSenseCustomerSupport

MartyG-RealSense, 2023-11-22:
Hi @denseishin  Which RealSense camera model and ROS1 wrapper version are you using, please?

denseishin, 2023-11-23:
I'm using a D405, but to do so I'm using rjwb1's fork https://github.com/IntelRealSense/realsense-ros/pull/2297

MartyG-RealSense, 2023-11-23:
Even though development on the official ROS1 wrapper has ceased, the RealSense ROS development team recently added support for the D405 PID to the wrapper, as described at https://github.com/IntelRealSense/realsense-ros/pull/2915