realsenseai/realsense-ros #904: Launch camera based on port number (or model)

doisyg, 2019-08-29:
Currently, the only way to distinguish which camera to use in a multi camera/multi model configuration is to use the serial number of the camera device. This is not easy to maintain.
Would it be possible to use the *camera1_usb_port_id* as in the legacy package?
https://github.com/IntelRealSense/realsense-ros/blob/ebc06d4be1800da82595a7ccb163f3b5b6e69b23/realsense_camera/launch/r200_nodelet_multiple_cameras.launch#L18-L26

In addition having a filter based on the camera model to be able to launch concurrently a D435 and a T265 without using serial numbers.

doronhi, 2019-10-24:
Added option in #966, #964, #960.

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

Hi @doisyg, Do you need further help on this?
If we don't hear from you in 7 days, we will close it.

closed 2020-03-20 by RealSenseCustomerSupport