realsenseai/realsense-ros #907: T265 enable_fisheye:=false incorrect error message

pringithub, 2019-08-29:
Heads up: when I run 
```
roslaunch realsense2_camera rs_t265.launch enable_fisheye:=false
```
The info stream shows that it is on, and topics are created. However, no data is being streamed (rqt_image_viewer shows nothing).

doronhi, 2019-09-02:
When I run the command you posted:
`roslaunch realsense2_camera rs_t265.launch enable_fisheye:=false` I see the topics and also the stream using `rqt_image_view`
The reason is that `enable_fisheye` parameter is a bug and the ones used are `enable_fisheye1` and `enable_fisheye2`. Setting them to false you get the expected results. 
I plan to fix the "ambiguity" together with the following issue regarding the [2 infrared sensors](https://github.com/IntelRealSense/realsense-ros/issues/872)

The question remained - why didn't you get the fisheye streams? Do you see them when the `enable_fisheye` flag is not set?
Do you see them using realsense-viewer?

doronhi, 2020-07-26:
Regarding my remark " plan to fix the "ambiguity"....", I would like to withdraw.
This is not a bug.
the parameter "enable_fisheye" refers to sensor fisheye with index 0 (exists on D430i).
Parameters "enable_fisheye1" and "enable_fisheye2" refer to the sensors available in TM2. "enable_infra" follows the same convention. It refers to "infra0" on L515 and in the D400 series you have infra1 and infra2.
Although it is true that you cannot enable just one fisheye sensor, you can, in the D400 series, enable just one infrared sensor.
One could argue that a parameter name "..._fisheye" or "..._infra" could refer to all parameters begin with that prefix.
It sounds right to me but not something that I see as very high on the list right now. I will be happy to consider a PR.

closed 2020-09-11 by RealSenseSupport