realsenseai/realsense-ros #908: T265 not working with RPi3B+

pringithub, 2019-08-29:
The T265 is not working with my RPi3B+.
It will recognize the device and say it is streaming, however `rqt_image_view` shows a blank screen. 
As well, the pose is accumulating heavy drift and is unusable. Perhaps this is a clock issue? Does the T265 latch onto the host computer's processing? I would think and hope not ... 
The T265 works perfectly on my beefed-up PC workstation.

Note this model only has USB2 ports. The RPI3B+ is able to process _most_ of the data coming out of my D435, however.

MartyG-RealSense, 2019-08-29:
Do you have the T265 and D435 connected at the same time, please?  There is a reported issue with this pairing, where one has to start up first before the other (my memory says that it is the D435 that should start first).

https://github.com/IntelRealSense/librealsense/issues/3488

https://github.com/IntelRealSense/librealsense/issues/3697

Also, if you are using a USB hub, it is recommended that the USB hub is powered up first and **then** the T265 is connected to the hub, otherwise the T265 may not be recognized if both hub and T265 try to activate at the same time.

pringithub, 2019-08-29:
@MartyG-RealSense what you are referencing is a separate issue I had just now :)
That issue occurred on my PC, where the D435 was left in and I started the T265 launch script. What happened there was the D435 logging info showed up; also the D435 topics were published, but inaccessible.

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Hi @pringithub

Reviewing your original issue with regards to the rqt_image_view showing blank and this being just for T265 connected to the system only.  Your note saying that the T265 is connected to a USB port, this is where things become clear to me.  The T265 is configured such as if it is connected to USB3 port then you're able to get full feature meaning:  left and right fisheye streaming to host platform, raw IMU streaming to host, 6DoF pose data to host.  If the T265 is connected to USB2 port, the only stream you're able to get to host platform is 6DoF pose data only.

The T265 is designed to process pose data on it's own ASIC and not on host directly.

Hope this helps.

Your original message:
The T265 is not working with my RPi3B+.
It will recognize the device and say it is streaming, however `rqt_image_view` shows a blank screen.
As well, the pose is accumulating heavy drift and is unusable. Perhaps this is a clock issue? Does the T265 latch onto the host computer's processing? I would think and hope not ...
The T265 works perfectly on my beefed-up PC workstation.
Note this model only has USB2 ports. The RPI3B+ is able to process _most_ of the data coming out of my D435, however.

closed 2019-10-11 by RealSenseCustomerSupport