realsenseai/realsense-ros #909: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info

InternTrossen, 2019-08-29:
Hi there, 

Specs:
OS: Ubuntu 16.04 LTS, Device: D435, USB3 connection, SDK: 2.18.1.

I am trying to get a point cloud with the following launch command.
`roslaunch realsense2_camera demo_pointcloud.launch 
`
The point cloud and camera image show up fine and I don't visually see any problem other than the following:

I seem to consistently get the error message shown in the title:
 `[ERROR] [1567112020.171953700]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
`
I am also consistently getting this warning:
`[ WARN] [1567110915.813420691]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
`

I have tried installing the SDK with [Debian Package](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) as well as from source and then following [this.](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)

Further, running `modinfo uvcvideo | grep "version:"` produces, `version:        1.1.1
srcversion:     FC28E0DFD6253AEC0D4A815
` - it does not include the realsense string as mentioned in the debian package installation guide.

Any and all help would be greatly appreciated, thanks in advance.

doronhi, 2019-09-02:
The first issue: `An error has occurred during frame callback` is most likely related to frames skipped due to momentarily or permanent cpu overload. It means that new frames arrived while previous ones were still being processed,.
The second issue: `Frame metadata isn't available!` is related to the problems in the kernel patch installation, just as the problem with `uvcvideo` you describe.
It's probably caused by installation both from source and from debian.
The best approach would be to start over, choosing one form of installation and sticking to it.
If you don't need access to the librealsense2 code, I recommend installing it from debian.

InternTrossen, 2019-09-03:
Hi @doronhi, 
Thanks for your response! I really appreciate the help!

The first issue: After doing a little more digging, it is most definitely a cpu overload, thanks!

The second issue: My apologies for the confusion but what I meant to say is that I had tried starting over from debian and from source, i.e. both attempts were from scratch. 

Currently I'm going with Debian:
After installing `librealsense2-dkms`, part of the output is `First Installation: checking all kernels...
Building only for 4.15.0-58-generic
Module build for the currently running kernel was skipped since the
kernel source for this kernel does not seem to be installed.`

It seems like somewhere/somehow I may have removed kernel files poorly?

doronhi, 2019-09-08:
You may need to install a specific kernel patch if you have kernel version 4.15. Checkout [librealsense issue #4669] (https://github.com/IntelRealSense/librealsense/issues/4669#issuecomment-522759949)

InternTrossen, 2019-09-09:
Hey @doronhi ,

Thanks for the help!
It seems [that kernel patch](https://github.com/IntelRealSense/librealsense/issues/4669#issuecomment-522759949) is specific to 4.15.0-58 and I have 4.15.0-60.
I re-booted with an earlier kernel version (4.15.0-45) and went through the install again, this time successfully.

closed 2019-09-09 by InternTrossen