realsenseai/realsense-ros #910: Jetson Nano | Error while using "catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release"

RatheeshRK, 2019-09-01:
Trying to use Realsense D435 on Jetson Nano
Installed Realsense SDK
We are able to use the D435 using the realsense-viewer (with out any error) from Terminal, on Ubuntu 18.04 LTS

Error while using "catkin_make -DCATKIN_ENABLE_TESTING=False `-DCMAKE_BUILD_TYPE=Release"`

`....@jetsonN:~/realsense_ros_ws$ catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
Base path: /home/rrk/realsense_ros_ws
Source space: /home/rrk/realsense_ros_ws/src
Build space: /home/rrk/realsense_ros_ws/build
Devel space: /home/rrk/realsense_ros_ws/devel
Install space: /home/rrk/realsense_ros_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/rrk/realsense_ros_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/rrk/realsense_ros_ws/build"
####
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  9%] Built target ddynamic_reconfigure
[  9%] Built target sensor_msgs_generate_messages_nodejs
[  9%] Built target _realsense2_camera_generate_messages_check_deps_Extrinsics
[  9%] Built target std_msgs_generate_messages_nodejs
[  9%] Built target std_msgs_generate_messages_eus
[  9%] Built target _realsense2_camera_generate_messages_check_deps_IMUInfo
[  9%] Built target sensor_msgs_generate_messages_eus
[  9%] Built target std_msgs_generate_messages_cpp
[  9%] Built target sensor_msgs_generate_messages_cpp
[  9%] Built target sensor_msgs_generate_messages_py
[  9%] Built target std_msgs_generate_messages_py
[  9%] Built target actionlib_generate_messages_cpp
[  9%] Built target _catkin_empty_exported_target
[  9%] Built target rosgraph_msgs_generate_messages_eus
[  9%] Built target rosgraph_msgs_generate_messages_lisp
[  9%] Built target dynamic_reconfigure_gencfg
[  9%] Built target rosgraph_msgs_generate_messages_cpp
[  9%] Built target dynamic_reconfigure_generate_messages_py
[  9%] Built target dynamic_reconfigure_generate_messages_nodejs
[  9%] Built target roscpp_generate_messages_cpp
[  9%] Built target dynamic_reconfigure_generate_messages_cpp
[  9%] Built target dynamic_reconfigure_generate_messages_lisp
[  9%] Built target roscpp_generate_messages_py
[  9%] Built target rosgraph_msgs_generate_messages_py
[  9%] Built target rosgraph_msgs_generate_messages_nodejs
[  9%] Built target dynamic_reconfigure_generate_messages_eus
[  9%] Built target roscpp_generate_messages_eus
[  9%] Built target roscpp_generate_messages_nodejs
[  9%] Built target roscpp_generate_messages_lisp
[  9%] Built target diagnostic_msgs_generate_messages_nodejs
[  9%] Built target diagnostic_msgs_generate_messages_lisp
[  9%] Built target diagnostic_msgs_generate_messages_eus
[  9%] Built target tf2_msgs_generate_messages_nodejs
[  9%] Built target tf2_msgs_generate_messages_py
[  9%] Built target tf2_msgs_generate_messages_lisp
[  9%] Built target tf2_msgs_generate_messages_eus
[  9%] Built target tf2_msgs_generate_messages_cpp
[  9%] Built target actionlib_generate_messages_py
[  9%] Built target actionlib_generate_messages_nodejs
[  9%] Built target actionlib_generate_messages_eus
[  9%] Built target actionlib_msgs_generate_messages_py
[  9%] Built target bond_generate_messages_lisp
[  9%] Built target actionlib_msgs_generate_messages_nodejs
[  9%] Built target diagnostic_msgs_generate_messages_cpp
[  9%] Built target actionlib_msgs_generate_messages_lisp
[  9%] Built target geometry_msgs_generate_messages_py
[  9%] Built target geometry_msgs_generate_messages_nodejs
[  9%] Built target tf_generate_messages_lisp
[  9%] Built target nav_msgs_generate_messages_lisp
[  9%] Built target actionlib_msgs_generate_messages_eus
[... 835 lines cut ...]
/usr/local/include/librealsense2/hpp/rs_processing.hpp:204:11: note: candidate: rs2::pointcloud::pointcloud(rs2::pointcloud&&)
/usr/local/include/librealsense2/hpp/rs_processing.hpp:204:11: note:   candidate expects 1 argument, 2 provided
In file included from /usr/include/boost/config.hpp:61:0,
                 from /usr/include/boost/bind/bind.hpp:24,
                 from /usr/include/boost/bind.hpp:22,
                 from /opt/ros/melodic/include/class_loader/class_loader.hpp:35,
                 from /opt/ros/melodic/include/pluginlib/./class_list_macros.hpp:40,
                 from /opt/ros/melodic/include/pluginlib/class_list_macros.h:35,
                 from /home/rrk/realsense_ros_ws/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:6,
                 from /home/rrk/realsense_ros_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/rrk/realsense_ros_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/boost/function/function_template.hpp: At global scope:
/usr/include/boost/function/function_template.hpp:707:5: error: ‘boost::function1<R, T1>::function1(Functor, typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type) [with Functor = realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)::<lambda(bool)>; R = void; T0 = bool; typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type = int]’, declared using local type ‘realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)::<lambda(bool)>’, is used but never defined [-fpermissive]
     BOOST_FUNCTION_FUNCTION(Functor BOOST_FUNCTION_TARGET_FIX(const &) f
     ^
/usr/include/boost/function/function_template.hpp:707:5: error: ‘boost::function1<R, T1>::function1(Functor, typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type) [with Functor = realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)::<lambda(int)>; R = void; T0 = int; typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type = int]’, declared using local type ‘realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)::<lambda(int)>’, is used but never defined [-fpermissive]
/usr/include/boost/function/function_template.hpp:707:5: error: ‘boost::function1<R, T1>::function1(Functor, typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type) [with Functor = realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)::<lambda(double)>; R = void; T0 = double; typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type = int]’, declared using local type ‘realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)::<lambda(double)>’, is used but never defined [-fpermissive]
/usr/include/boost/function/function_template.hpp:707:5: error: ‘boost::function1<R, T1>::function1(Functor, typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type) [with Functor = realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)::<lambda(int)>; R = void; T0 = int; typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type = int]’, declared using local type ‘realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)::<lambda(int)>’, is used but never defined [-fpermissive]
realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
CMakeFiles/Makefile2:2076: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@RatheeshRK What's the SDK version and ROS wrapper version when you got the issue? Did you still have the issue with the latest version? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@RatheeshRK  Sorry that we didn't get response from you for weeks. Will close it at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2019-12-13 by RealSenseCustomerSupport

alexCharters, 2022-03-09:
I had the same issue. To solve, I went and manually edited the base_realsense_node.cpp file. All you have to do is preface each instance of find_if with std::