realsenseai/realsense-ros #911: Is there any official guidance to simulate D435i in gazebo&RViz?

JackStig, 2019-09-02:
Recently I'm working on detecting obstacle by simulating D435i in gazebo and visualizing it on RViz. I've found some useful file on realsense2_description but I have not found overall guidance on it.Now I'm stuck on adding D435i to gazebo,so I want to read some practical urdf file(especially the one simulating with robot arm). Any help will be grateful,thank you.

MartyG-RealSense, 2019-09-02:
Intel published a Gazebo plugin for RealSense, though it is a few years old.  Looking in the files, it seems that it was intended to simulate the older RealSense R200 stereo camera model.

https://github.com/intel/gazebo-realsense

A RealSense user created their own evolution of this plugin that contains instructions for using it with RViz.

https://github.com/SyrianSpock/realsense_gazebo_plugin

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

Hi @JackStig,

Any update? I hope the info provided by @MartyG-RealSense resolves your question.

JackStig, 2019-11-07:
Hi @MartyG-RealSense @RealSenseCustomerSupport ,
Thanks for your concern,sorry I was too busy to reply you.I find an official annoucement saying that there is no plugin for RViz&Gazebo,while repos in Git are too old to use.So I just give up and devote on something else.

MartyG-RealSense, 2019-11-07:
@JackStig   Thanks so much for your update.  I am sorry that you did not find what you needed.

closed 2019-11-07 by RealSenseCustomerSupport

sayyidabeegam, 2020-10-16:
Hello
I want to use D435 with my robot manipulator.
I have gone through above links.
But I don't get any idea to integrate the sensor with my manipulator.
My requirements are
**1. I want to attach the D435 with my manipulator's base link
2. want to launch with my manipulator package
3. want to use all available sensor data format.**
Please help me to figure it out..
Thanks in advance

MartyG-RealSense, 2020-10-16:
Hi @sayyidabeegam   I have responded to your question at https://github.com/IntelRealSense/realsense-ros/issues/1453#issuecomment-710226770 - thanks!

rickstaa, 2021-12-29:
I will also put this here for future reference. An example of how to use the D435i camera in Gazebo can be found [here](https://github.com/rickstaa/realsense-ros-gazebo), might people need it. A detailed explanation video created by [@issaiass](https://github.com/issaiass) can  be found [here](https://www.youtube.com/watch?v=hpUCG6K5muI&feature=youtu.be).