realsenseai/realsense-ros #914: depth accurary of d435i

Garand0o0, 2019-09-04:
I found that the depth accuracy of d435i is not good. And the dynamic calibration cannot improve it very well, I want to make a comparison table to improve the quality of the depth. If I have a comparison table now, where should I put it in. Thanks!

MartyG-RealSense, 2019-09-04:
There are a range of factors (including environmental factors and distance of the observed object from the camera) that can affect the depth accuracy of a 400 Series camera, and methods for improving depth quality.  Intel's camera tuning guide is an excellent all-in-one document for achieving better results.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

Page 3 of this guide contains a formula for plotting out 'RMS error' (how depth noise will increase as an observed object gets further from the camera).  The top of page 4 has a pre-made graph showing how RMS may increase over distance with a D435 model compared to the D415 model.  As suggested by the curves, the D415 can provide around 2.5x the image quality of the D435 due to the hardware differences in their sensors.

If you plot your own chart, you could insert it into the comments here with a copy-and-paste of an image of the chart.

Garand0o0, 2019-09-04:
Hi MartyG-realsense, thanks for you reply. If I plot my own chart or have a table or a formula, what should I do? Should I change the code? Which code should I modify? I don't understand "you could insert it into the comments here with a copy-and-paste of an image of the chart. The paste will automatically generate the image in the comments without having to insert a file.".I would appreciate it if you could tell me some details.

MartyG-RealSense, 2019-09-04:
I apologize for any confusion.  I thought you were asking for a way to visually share your comparison table results in the comments so that others can see it.  

If this is what you want to do, you could take a screen-capture of whatever is currently displayed on your computer screen (e.g with the 'Print Screen' key on a PC).  If you then do a paste (the keyboard shortcut CTRL-V on a PC) whilst in the comments box then it will paste the image of what you captured from your screen into your comment and display it, like I did with my curve graph above.

If this is not what you meant, please provide more information about what you are trying to do.   Thanks!

RealSenseCustomerSupport, 2019-11-02:
----------------------------------------------

Please let us know if you still need further help on this ticket. Thank you.

closed 2019-11-14 by RealSenseCustomerSupport