realsenseai/realsense-ros #915: occupancy-mapping and camera position

jungladicitta, 2019-09-04:
I have mobile robot based on turtlebot3 waffle but with mast 1.6 meters height.
I am trying to build a map with using t265 and d435 cameras together.
To do that i am using occupancy package from occupancy-mapping branch.
So, i have two questions, and they are all about using occupancy-mapping branch.

I. first one is what is the best position of cameras on tall robot with 1.6 meters height for better mapping? should i 
    1) place cameras in the middle of the robot aimed straight
[image]
or
    2) place cameras on top of the mast aimed to the floor
[image]

II. I have problem with tf. I have placed camera somewhere on the mast and I am running occupancy. The problem is that I didnt configure frames correctly, so map is located in wrong position, near camera position instead of floor.
[image]

How can i configure frames correctly?
my launch file with static_transform_publishers:

```
<launch>

  <!-- CAMERA -->
  <include file="$(find turtlebot3_occupancy_mapping)/launch/robot_and_cameras.launch"/>

  <!-- occupancy-mapping -->
  <include file="$(find occupancy)/launch/occupancy.launch">
  </include>

  <node pkg="turtlebot3_occupancy_mapping" type="map_frame_publisher.py" name="map_frame_publisher" output="screen">
  </node>

  <node name="static_tf0" pkg="tf2_ros" type="static_transform_publisher" args="0 0 1.3 0 0 0 /map /t265_odom_frame"/>

  <node name="static_tf1" pkg="tf2_ros" type="static_transform_publisher" args="0 0 0 0 0 0 /map /odom"/>

  <!-- RVIZ -->
  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find turtlebot3_occupancy_mapping)/rviz/pointcloud_w_occupancy.rviz">
  </node>

</launch>
```

piece of my urfd.xacro file:
```
  <joint name="cameras_joint" type="fixed">
    <parent link="base_footprint"/>
    <child link="t265_pose_frame"/>
    <origin xyz="0.0 0.0 1.3" rpy="0.0 0.0 0.0"/>
  </joint>

  <link name="t265_pose_frame"/>
  <link name="d400_link"/>

  <joint name="t265_to_d400" type="fixed">
    <parent link="t265_pose_frame"/>
    <child link="d400_link"/>
    <origin xyz="0.009 0.021 0.027" rpy="0.000 -0.018 0.005"/>
  </joint>
```

my map frame publisher script, which i use to publish "/occupancy" topic to "/map" frame:
```
#!/usr/bin/env python  
import rospy

# Because of transformations
import tf_conversions

import tf2_ros
import tf2_msgs.msg
import geometry_msgs.msg
import sensor_msgs.msg
import nav_msgs.msg

def handle_map_frame(data):
    pub = rospy.Publisher("map", nav_msgs.msg.OccupancyGrid, queue_size=10)
    rate = rospy.Rate(10) # 10hz
    pub.publish(data)
    rate.sleep()

if __name__ == '__main__':
    rospy.init_node('map_frame_broadcaster')
    rospy.Subscriber("/occupancy", nav_msgs.msg.OccupancyGrid, handle_map_frame)
    ##rospy.Subscriber('/sensor_us', sensor_msgs.msg.Range, handle_camera_pose)
    rospy.spin()
```
What can i do to draw map on the floor?

jungladicitta, 2019-09-05:
II-nd question is solved.
I forgot to indicate frame for occupancy data:
`data.header.frame_id = "/map"`

jungladicitta, 2019-09-06:
about I-st question.

I have tested second case where cameras is located on the top of the mast looking towards floor. I found out few features.

1) HeightOfInterestMax - is the distance from camera to the floor;
    HeightOfInterestMin -                                             to the top of the mast (height of my robot).
These parameters are from occupancy.launch file.
```
<launch>
  <param name="resolution" value="0.03"/>
  <param name="HeightOfInterestMin" value="-0.37"/>
  <param name="HeightOfInterestMax" value="1.27"/>
  <param name="DepthOfInterestMin" value="0.2"/>
  <param name="DepthOfInterestMax" value="2.0"/>

  <node name="occupancy_node" pkg="occupancy" type="occupancy_node" output="screen">
  </node>
</launch>
```

2) every time when I move cameras on the mast I need to set HeightOfInterestMax parameter precisely. If I set this value too large - floor become obstacle and so all map turns to black, too small - I cant detect small obstacles lying on the floor.
Actual height of cameras is around 1.32 meters:
[image]
but HeightOfInterestMax needs to be set to 1.27 otherwise floor become obstacle.

So I put these two boxes on the floor:
[image]

and i got this map:
[image]

closed 2019-09-06 by jungladicitta

rjam92, 2020-10-17:
hi, can you show me what your tf tree finally looked like?