realsenseai/realsense-ros #916: T265 camera crashes on raspberry pi

jcgarciaca, 2019-09-04:
I am getting an error with using T265 camera on raspberry pi 3. It is, when I run the command `roslaunch realsense2_camera rs_t265.launch`, after some minutes the error below appears. Sometimes the error appears soon (just a couple of minutes) and sometimes it takes around one or two hours to appear.
```
19:57:13.679 [2601] [E] Device-63B0: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
19:57:13.932 [2592] [E] Device-63B0: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
19:57:13.932 [2592] [E] Device-63B0: Entered state [ERROR_STATE]
```
I changed the camera (multiple T265) as well as the raspberry, but the error appears anyway.
I also updated the `LibRealSense` version such that I tested with `v2.23.0`, `v2.24.0` and `v2.27.0`. But same error still appears.

How could I solve this issue?

RealSenseCustomerSupport, 2019-10-18:
----------------------------------------------

@jcgarciaca - Typically these errors that you're seeing have to do with USB connection issues.  Having or making sure the T265 is connected via USB3 (pose data, fisheye data, IMU data can all be transferred to host when connected via USB3) or USB2 (pose data only can be transferred to host when connected via USB2).  If the device is not enumerated at the host in one of these two states, please do check your cable and check whether or not you are connecting via a USB hub.  The hub must be powered on and enumerated before T265 is connected or provided power.

Also the datasheet can be found here:  https://dev.intelrealsense.com/docs/tracking-camera-t265-datasheet

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@jcgarciaca - Hi, checking in to see if you're able to look into the notes provided.  Also have you attempted to run a sample from librealsense directly instead of going through the ROS launch?

closed 2019-12-05 by RealSenseCustomerSupport